trezor/trezor-firmware #990: make sure we have a replacement for test_zerosig

matejcik, 2020-05-13:
[test_zerosig.py](https://github.com/trezor/trezor-firmware/blob/b3569a53e6cb48e0f3874eaa32cef328d0d24602/tests/device_tests/test_zerosig.py) seems to be testing that particular Bitcoin transaction signatures have a leading zero stripped.

The test cases are invalidated in #959 because they are using invalid BIP32 paths. I have no way to regenerate the appropriate test vectors, because I'm not even completely sure what is the condition being tested.
In any case, it _seems_ that this is a feature that should be tested on a lower level, probably in trezor-crypto.

tsusanka, 2020-05-21:
Let's indeed move this to trezor-crypto. Maybe it is present already?

closed 2020-05-21 by tsusanka

prusnak, 2020-05-21:
These are the tests (last 4 usecases): https://github.com/trezor/trezor-firmware/blob/6fafe4c9ee7a423c0f7ed05dfb2186dd1efaa0df/core/tests/test_trezor.crypto.der.py#L29-L43