trezor/trezor-firmware #991: Collect all English strings into a single JSON/XML

tsusanka, 2020-05-14:
After #989 is implemented we should be able to collect all the English strings into a single file. This file will be then fed into [Crowdin](https://crowdin.com/) for the actual translating process.

We are currently considering two options:

1. Extract the strings using libcst/gettext, see @matejcik's comment [here](https://github.com/trezor/trezor-firmware/issues/989#issuecomment-627883458). In that case this issue is slightly duplicate to #989 and probably should be done together.
2. Collect the strings using device tests. Similarly to UI tests, we would run device tests against the emulator and gather all the strings that are displayed. Each time something is printed on the display we would store the string in some local json/xml file. Later we would concatenate all those files into a single one.

Both should be done automatically on CI.

---

Needs #989. Part of #987.

prusnak, 2020-05-14:
I think this task is not really blocked by fully implementing #989. We can collect all localized strings to a separate file (e.g. `i18n.py`) before the blob mechanism is in place.

matejcik, 2020-05-14:
Agreed. We need i18n.py anyway, the emulator-running-from-sources must be able to load strings without using a blob.

tsusanka, 2020-05-21:
> We can collect all localized strings to a separate file (e.g. i18n.py) before the blob mechanism is in place.

Let's do this first!

tsusanka, 2020-05-26:
I have merged #989 into this one.

closed 2024-02-12 by grdddj