trezor/trezor-firmware #993: Create a simple webapp mimicking Trezor screen

tsusanka, 2020-05-14:
During our meeting on #987 we have realized context for the translators is quite important, but what seems as even more important for the translators is to see how the translated text looks like in "a real world" - on Trezor. The translator might need to think about what words suit the screen the best and this would help a lot.

The idea is to create a simple webapp with Trezor screen where the text is editable. Ideally of course both T1 and TT are supported.

---
Part of #987.
To be discussed how much we need this.

tsusanka, 2020-05-21:
@prusnak has an idea which he would like to try out :).

prusnak, 2020-06-04:
Added in d9d5511858bcd7958186c5e1516cc1c1310c8502

It is not pixel-perfect, but more than usable for our purposes.

closed 2020-06-04 by prusnak

tsusanka, 2020-06-04:
This is great! :tada: