trezor/trezor-firmware #994: Translate bootloader

tsusanka, 2020-05-14:
From the product point of view it would be really nice to have the bootloader in native language as well because it is needed for the firmware upgrade. Since the firmware upgrade might be stressful for many of our users already, failing to provide a native interface seems as a bummer.

It seems to me we currently have number of ideas:
1. Do it simply and hard-code the translated strings into the bootloader directly. This doesn't scale, but might be a quick working solution for the time-being. We would support only ~5 main languages for example.
2. Use the very same mechanism as described in the firmware. We would have to be careful here not to introduce another attack vendor; the bootloader should be as simple as it gets after all.
3. Not do it and come up with another product solution (replace with icons, provide a great guide in Suite etc.).

prusnak, 2020-05-14:
1) is kind of OK, but only for Model T, for Model 1 we can hardly add anything into the bootloader

2) is a NO GO

Option 3) is the best IMO

Hannsek, 2023-03-17:
Can we close this one? I don't think we want to add multiple languages into bootloader.

closed 2023-03-17 by Hannsek

My1, 2023-03-17:
agree multiple BL Languages opens a can of worms especially as unicode might need to be introduce Unicode with its own problems.