trezor/trezor-firmware #999: Retrieve the current master fingerprint

achow101, 2020-05-16:
PSBTs use the master key's fingerprint to identify signers. It seems that a recent change to Trezor T makes it no longer possible to retrieve the fingerprint. Previously, I was able to do this by requesting the key at `m/0'` and extracting the parent fingerprint from that. But doing that now results in `DataError: Forbidden key path`. Presumably this is because `m/0'` is not a standard derivation path so this is disallowed. If it is intended that users cannot request keys at non-standard derivation paths, then some other way to retrieve the master key fingerprint is needed to work with PSBT. This could be just a new message type that returns the fingerprint.

tsusanka, 2020-05-18:
This happens on master not on 2.3.0 right?

matejcik, 2020-05-18:
I don't think a special message is necessary, more like we need to add an exception for this case. Something like `coin_name=="Bitcoin" and script_type is None and address_n == [0 | HARDENED]`?

Perhaps we could just add `m/0'` prefix to the list of allowed namespaces.

matejcik, 2020-05-18:
FWIW, i'm not even sure if the protections should be enabled for GetPublicKey.
Short of breaking the public-key crypto, can we imagine an attack that could abuse GetPublicKey on an unapproved path?

destrys, 2020-05-18:
Agreed on not limiting access to BIP32 spaces.

But I do think it would be really nice to have the master key's fingerprint available without having to pull `m/x'`. Returning the XFP on every call of `getPublicKey` would be very useful. It's fewer calls when exporting an xpub, and it would allow a user to not have to expose an xpub at depth 1.

commit 2f665c8f (2020-05-21): core: add exception to GetPublicKey for PSBT master fingerprint (fixes #999)
closed 2020-05-21 by matejcik