trezor/trezor-firmware #1000: FIDO2 on Trezor One

1000i100, 2020-05-18:
U2F authentication is check in your website for Trezor One but not FIDO2.
Is there hardware limit about this ?

I would like to use FIDO2 on a cheaper device than Model T.

There is some word about it in #39 after it have been closed.

prusnak, 2020-05-18:
Currently, FIDO2 is supported only on Trezor Model T.

My1, 2020-07-11:
a better answer than in 2019 when it was not even planned, but still kinda sad it's just on the (kinda expensive) model T

m5x, 2021-03-08:
Is FIDO2 still coming to Trezor One? It would be great to have it, Trezor T is too bulky for my usage.

prusnak, 2021-03-08:
> Is FIDO2 still coming to Trezor One? It would be great to have it, Trezor T is too bulky for my usage.

Yes, but no ETA yet

aasami, 2022-02-09:
Another year passed. Still no support? This feels like hydrogen fuel adoption. Forever the fuel of the future.

My1, 2023-01-01:
I guess it's 2023 now. would be really cool for having FIDO2 on The T1.

Jordo0G, 2024-01-04:
Any update? 2024 and Trezor One is still a great device. I know the Safe 3 is now the flagship, but there's so many One's out there just waiting to be used for FIDO2.

prusnak, 2024-01-04:
FIDO2 is not coming to Trezor One. Get a Trezor Safe 3 if you want FIDO2 on smaller device than Trezor Model T.

closed 2024-01-04 by hynek-jina