trezor/trezor-firmware #1006: Translate wire errors

tsusanka, 2020-05-21:
We send an informative message when a wire error occurs. This message needs to be translated as well.

We can:
1. Simple translate these using the very same translation mechanism we are developing.
1. Add an id for each wire error. Store the messages separately. That means `Failure` would have `id` instead of `message`.

Brought up by @szymonlesisz during discussion. However, at the moment I can't remember what is wrong about point 1. Ideas?

---

Part of #987.

prusnak, 2020-05-21:
FTR `Failure` already has `code` enum field. I  guess it would be good to stop using the `message` field and focus on making the `code` field more useful (i.e. add more types).

```
message Failure {
    optional FailureType code = 1;  // computer-readable definition of the error state
    optional string message = 2;    // human-readable message of the error state
    enum FailureType {
        Failure_UnexpectedMessage = 1;
        Failure_ButtonExpected = 2;
        Failure_DataError = 3;
        Failure_ActionCancelled = 4;
        Failure_PinExpected = 5;
        Failure_PinCancelled = 6;
        Failure_PinInvalid = 7;
        Failure_InvalidSignature = 8;
        Failure_ProcessError = 9;
        Failure_NotEnoughFunds = 10;
        Failure_NotInitialized = 11;
        Failure_PinMismatch = 12;
        Failure_WipeCodeMismatch = 13;
        Failure_FirmwareError = 99;
    }
}

tsusanka, 2020-05-21:
> FTR Failure already has code enum field. I guess it would be good to stop using the message field and focus on making the code field more useful (i.e. add more types).

Yes, indeed. The current situation is however we have ProcessError and DataError at most places and we use the message to particularize the problem. I will add this as an option 3. I believe that in such case the Suite should stop showing the message and fetch it from somewhere else (translated). The message field would stay to serve third parties.

matejcik, 2020-05-22:
I'm not sure if it makes sense to enumerate all possible errors as codes.

For instance, in Bitcoin field validation, there is an error "{field_name} is not allowed on this coin" for about four different fields. ISTM having four separate enum codes for this is pointless.

It *might* make sense to enumerate different reasons for failing to sign a Bitcoin transaction -- but OTOH we already fold different kinds of internal check failures into the same "Transaction was changed during signing" string.

Also looking at the list of codes, that could be pared down. It's unclear to me why we have `PinExpected` and `ButtonExpected` instead of using the generic `UnexpectedMessage`; and `PinCancelled` separate from `ActionCancelled`.

This looks like two tasks:
(1) improve granularity of error codes
(2) translate the accompanying strings anyway, using the normal translation mechanism

tsusanka, 2020-06-04:
Let's translate the wire error strings directly in firmware.

closed 2020-06-04 by tsusanka