trezor/trezor-firmware #1007: Reuse derived Shamir seed in Cardano app

matejcik, 2020-05-22:
With the introduction of session cache, we stopped caching plaintext of the passphrase. That means that with Single backup (BIP39), we need to ask for passphrase separately in the Cardano app -- because Cardano derives the root from the mnemonic via a different algorithm than BIP39.

This is no longer true with SLIP39+SLIP23: the SLIP39 Master Secret (which already incorporates the passphrase) is used for deriving the Cardano root directly.

Our code does not take advantage of this; instead, it asks the user for a passphrase in all cases, and re-derives the Master Secret.

In case the device is using Shamir backup, we should reuse the cached Master Secret instead of prompting for passphrase separately.

closed 2020-09-01 by tsusanka

tsusanka, 2020-09-25:
## QA

Please try Cardano signing at adalite.io. Notice you exceptionally need to use master (59b8a5a) not what we have frozen.

tsusanka, 2020-12-28:
Cardano signing tested as part of _Freeze_. See [testpad](https://satoshilabs.ontestpad.com/script/266#6/3/).

@bosomt could you please test one more thing here and that's:
- Take latest FW 2.3.4.
- Recover Trezor with some mnemonic (whatever).
- Use some passphrase (whatever).
- Get Cardano address via adalite.
- Do the very same thing with 2.3.5 RC, same mnemonic, same passphrase.
- Verify the addresses match.

Thanks!

bosomt, 2020-12-28:
@tsusanka ON IT

bosomt, 2020-12-28:
i see tx history, addresses and funds with same mnemonic and passphrase

**regular-2.3.5-889a3699**

[image]

**public 2.3.4**

[image]

bosomt, 2020-12-28:
and now wish Shamir seed :)

**2.3.4**

addr1qydjh946sw6d83tg5v6x0z8f0zayd68xpjefwxv9ulf7ke76fr6d4ccs5gfn4aduwg8f2dx370m2gynhlzdzkcnrlwzs0x9hja

[image]

**regular-2.3.5-889a3699**

addr1qydjh946sw6d83tg5v6x0z8f0zayd68xpjefwxv9ulf7ke76fr6d4ccs5gfn4aduwg8f2dx370m2gynhlzdzkcnrlwzs0x9hja

[image]

tsusanka, 2020-12-28:
Great, thank you!

commit c96e5780 (2020-06-03): core/cardano: reuse derived Shamir seed (fixes #1007)
commit 7682383f (2020-07-10): fixup! core/cardano: reuse derived Shamir seed (fixes #1007)
commit ebb07ed6 (2020-07-10): core/cardano: reuse derived Shamir seed (fixes #1007)
commit f6e9f04d (2020-07-14): core/cardano: reuse derived Shamir seed (fixes #1007)