trezor/trezor-firmware #1008: 2.3.1 RC1 doesn't boot up

bosomt, 2020-05-23:
**Describe the bug**
RC1 of 2.3.1 doesn't boot up

**Firmware version or revision**
2.3.1 regular and 2.3.1 btc-only won't start 
tested on two devices 

**Desktop/smartphone setup (please complete the following information):**
nixOS

**To Reproduce**
Steps to reproduce the behavior:
1) load fw to device
2) confirm that it's not signed
3) see black screen

**Expected behavior**
device will boot up

**Screenshots**
https://photos.app.goo.gl/NTATmxRWP8twSgna6

**Additional context**
Add any other context about the problem here.

tsusanka, 2020-05-23:
@bosomt please always include the firmware revision. It is 2dc0107b correct?

tsusanka, 2020-05-23:
I am confirming this. The device is also not listed via `trezorctl list`.

bosomt, 2020-05-23:
sorry I forgot
that's correct

ls | grep 2.3.1
trezor-fw-btconly-2.3.1-2dc0107b.bin
trezor-fw-regular-2.3.1-2dc0107b.bin

tsusanka, 2020-05-23:
Happens on current master (675ff459) as well.

tsusanka, 2020-05-23:
Bumping priority in https://github.com/trezor/trezor-firmware/issues/946. This is exactly what we would have caught early if it would be implemented.

tsusanka, 2020-05-23:
I have bisected this to 0d65d684f0bb2d4c604c0a7822237b5095f74bd5. That is the first bad commit.

Maybe the TREZOR_MODEL variable is not properly set to "T"? I have tried to build with `TREZOR_MODEL=T make build_firmware` with no success though. I think `make build_firmware` still should build Trezor T firmware on default.

@hiviah could you look at this please? Of course on Monday is enough.

hiviah, 2020-05-23:
This was caused by weird gcc property where .s assembler files are not preprocessed and .S files are, so T1 code was getting mixed into TT code. Fix in pull request #1009 .

prusnak, 2020-05-24:
Closed via https://github.com/trezor/trezor-firmware/pull/1009

closed 2020-05-24 by prusnak