trezor/trezor-firmware #1011: Forbidden key path on wallet.trezor.io (2.3.1 RC2)

tsusanka, 2020-05-25:
On wallet.trezor.io basically nothing loads up due to error `Error details: Forbidden key path`.

I am afraid Wallet is using a lot of old paths.

tsusanka, 2020-05-25:
Related: https://github.com/satoshilabs/mytrezor/issues/720. I though it was fixed, but it most likely wasn't :(.

prusnak, 2020-05-25:
Ideally we should fix satoshilabs/mytrezor#720.

matejcik, 2020-05-25:
or we can punch in a hole for `m/0` same as we have for `m/0'`. It's the same thing after all, isn't it?

tsusanka, 2020-05-25:
> Ideally we should fix satoshilabs/mytrezor#720.

Although this seems like the correct solution, Suite team has other priorities.

> or we can punch in a hole for m/0 same as we have for m/0'. It's the same thing after all, isn't it?

This and removing the hotfix after killing old Wallet seems reasonable.

---

I will try out @matejcik's hotfix and talk to @mroz22 how difficult the change would be in Wallet.

matejcik, 2020-05-26:
wallet is actually using path `m/44/1/0` (non-hardened). We can of course add an exception, but waiting for this to be fixed in mytrezor seems a more correct course of action.

bosomt, 2020-05-26:
same result with Electrum

[image]

bosomt, 2020-05-26:
or exodus

[image]

matejcik, 2020-05-26:
confirmed in Electrum

matejcik, 2020-05-26:
Electrum is using an empty path to get the root XPUB.

matejcik, 2020-05-26:
Exodus is querying the following paths, all via `GetPublicKey` and with `coin_name=None`:

```
m/44h/0h/0h (BTC)
m/49h/0h/0h (BTC)
m/44h/60h/0h (Ethereum)
m/44h/5h/0h  (Dash)
m/44h/42h/0h  (Decred)
m/49h/156h/0h  (BTG)
m/49h/2h/0h  (LTC)
m/44h/145h/0h (Bcash)
m/44h/61h/0h  (ETC)
m/44h/133h/0h  (Zcash)
m/44h/148h/0h (ed25519) (Stellar)
m/49h/20h/0h  (Digibyte)
m/44h/144h/0h  (Ripple)
m/44h/3h/0h (Doge)
m/44h/1729h/0h (ed25519)  (Tezos)
m/44h/134h/0h (Lisk)
m/44h/175h/0h  (Ravencoin)
```
All of them seem to correspond to a valid coin. `ecdsa_curve_name` is mostly unspecified (therefore `secp256k1`), in the appropriate cases it is `ed25519`

IOW, all calls are valid security-wise. I'm not sure what to make of this. We *could* tell Exodus to fix their calls, but it seems pointless.

matejcik, 2020-05-26:
electrum does _not_ in fact need an empty path for the root xpub; instead, it needs the xpub at custom path:
```python
def get_derivation_used_for_hw_device_encryption():
    return ("m"
            "/4541509'"      # ascii 'ELE'  as decimal ("BIP43 purpose")
            "/1112098098'")  # ascii 'BIE2' as decimal
```

PR #1020 (2020-05-27): core: relax path checks on GetPublicKey
fixes #1011 by removing all restrictions on path for `GetPublicKey`, thus reverting to pre-#959 levels

this seems necessary due to the sheer amount of different uses of GetPublicKey:

- in trezor-agent, SLIP-13 and SLIP-17 paths are used
- in hwi, m/0' (and m/0 due to a bug) are used to retrieve master fingerprint
- in electrum, m/ELE'/something is used to derive a password for wallet file encryption
- in mytrezor wallet, m/44/1/0 path is used to check, uh, something
- in Exodus, `GetPublicKey` is used with valid paths, but without specifying `coin_name [... cut]
closed 2020-05-27 by tsusanka