trezor/trezor-firmware #1015: Make Core storage work on T1

tsusanka, 2020-05-26:
As discussed with @prusnak another issue to handle in #24 is to port Core's storage. Storage is a standalone project (see `storage` folder) that runs in Legacy as well, however here we mean to make `core/src/storage` work on T1.

In particular we would like to:
- make trezorconfig module work on T1
- make subset of `core/src/storage/` work on T1:
  - \_\_init.py\_\_
  - cache.py
  - device.py
  - residential_cred.py (nice to have)

We are not completely sure what this involves, we might need to change the memory layout of T1, modify bootloader etc.

It seems fine to copy&paste things to `src1` folder at the moment to avoid lot of `ifdef` statements and similar. We will have to tackle that later though.

prusnak, 2020-05-26:
Quickly looking into modtrezorconfig; it seems norcow config already respects TT/T1 memory layouts:

https://github.com/trezor/trezor-firmware/blob/d59ffe553f5bad7f1106a9dd163b718e15fd09a9/core/embed/extmod/modtrezorconfig/norcow_config.h#L28-L44

So there is a chance modtrezorconfig already works as is on T1.

tsusanka, 2020-06-15:
@hiviah this was done by #1031 right?

hiviah, 2020-06-15:
Yes.

closed 2020-06-15 by prusnak