trezor/trezor-firmware #1016: Figure out translations data blob signing process

tsusanka, 2020-05-26:
In #988 we are implementing the translations data blob. In this issue we should figure out how to sign the blob.

- Are we going to use another key than what is used to sign the firmware? Probably yes.
- Should this be auto-signed on Gitlab? Probably no.
- We might come up with a similar mechanism as firmware has but lower the signature threshold. In another words, only one of @prusnak / @slush0 is required to generate a valid signature.

---

Related #988. Part of #987.

Hannsek, 2023-11-20:
We will sign it as the eth definitions, right? @matejcik 
I believe we can close this.

matejcik, 2024-01-25:
yup, this is basically done as part of #3206 / #3491

closed 2024-01-25 by matejcik