trezor/trezor-firmware #1021: Clear cache after passphrase is turned on/off?

tsusanka, 2020-05-26:
@szymonlesisz reported this issue and I feel we have discussed it already but since I can't find any evidence of this I am filing a new issue.

Suppose this communication:

```
-> Intiliaze(state=None)
  <- Features

-> GetAddress
  <- PassphraseReq

-> PassphraseAck
  <- Address

..turn off passphrase here..

-> GetAddress
  <- Address  // the same address, the previous passphrase was used because it is cached
```

This seems a bit buggy but I am not sure there is a good solution. If we decide to clear the internal cache after the passphrase was turned off, we return `Address` corresponding to an empty passphrase which seems quite weird as well.

In other words, the Host should send Initialize after turning the passphrase off and failing to do so can result in weird consequences because we do not have sessions.

matejcik, 2020-05-27:
there is in fact [a testcase](https://github.com/trezor/trezor-firmware/blob/master/tests/device_tests/test_session_id_and_passphrase.py#L210-L230) for this exact behavior. I have a vague notion that we talked about it in the respective PR, but can't find it.

I _think_ the current behavior is technically correct. The `ApplySettings` call does not touch session management, the only calls to do that are `Initialize` and `WipeDevice` (plus `EndSession` in the future). You can change settings that affect how sessions are *created*, but there is no good reason to apply them "retroactively".

Of course if this is causing problems in applications, we can revisit this with new requirements?

matejcik, 2020-05-27:
FWIW we talked several times about some sort of `UnlockPassphrase` call that would just ensure the passphrase is cached. There's an argument to be made that this hypothetical command could do this "session reset" after the settings have been changed; or possibly `UnlockPassphrase(reset=true)` for that...

In all cases we reverted to (a) `Initialize` to do the reset part, and (b) `GetAddress` on a fixed path to do the unlock part.

tsusanka, 2020-05-27:
> there is in fact a testcase for this exact behavior. I have a vague notion that we talked about it in the respective PR, but can't find it.

> I think the current behavior is technically correct. The ApplySettings call does not touch session management, the only calls to do that are Initialize and WipeDevice (plus EndSession in the [...]

I agree with all of this. One of mine main motivations to open this issue was to document it.

> Of course if this is causing problems in applications, we can revisit this with new requirements?

@szymonlesisz Do you have some particular problems with this or it just surprised you?

szymonlesisz, 2020-05-28:
I just wanted a confirmation that it's a expected and correct behaviour because yes, it was a little surprising and at first sight i was blaming my code for doing this

tsusanka, 2020-05-28:
Ok, so I believe there is no further action required. At least it is documented here in case we run into it again. I might document it to `docs` as well, but is seems a bit arbitrary for that.

closed 2020-05-28 by tsusanka

matejcik, 2020-05-29:
> You can change settings that affect how sessions are _created_

just to raise a thing I realized re this little comment:

The following two sessions will behave differently:
```
> Initialize
> ApplySettings(enable passphrase)
> GetAddresss
< Address (with passphrase)
```
and
```
> Initialize
> GetAddress
< Address (without passphrase)
> ApplySettings(enable passphrase)
```

The former is a "session with passphrase" because the setting was applied on an empty cache. The latter is a "session without passphrase" because the setting was applied while the cache was already filled.

So it's not a property of session creation, but rather of the cache state.
I personally find this somewhat problematic, because the coupling between sessions and cache state is much looser than it (IMHO) should be.
But I currently don't have any idea how to make this better.

Maybe just to say that you're supposed to "unlock" after Initialize, and if you don't then it's your problem?