trezor/trezor-firmware #1022: Remove VeChain from supported coins

cryptomar1o, 2020-05-26:
VeChain did launch a new main net token that is no longer using ERC20 token standard. VeChain team decided to use the same address format that is used by Ethereum and because of that users and 3rd party services can easily mistake the VeChain mainnet token address with Ethereum address.

VEN on Ethereum network as ERC20 token still exists and it is supported by both Trezor's firmware, but users are confusing it with VET on VeChain Thor network (no application compatible with Trezor device exists) and losing access to their tokens.

VeChain (https://trezor.io/coins/#VEN) should be removed from supported coins & tokens that there is no more confusion for Trezor's users.

closed 2020-05-29 by matejcik

Homers1234, 2020-06-26:
Hi there. I was wondering if anyone could help me please.

I sent VeChain (VET) from Binance to the Trezor T Model (via MyEtherWallet) the Public Address: 
(0x8a1fa4811bD58DB02C6029BA63Dd17874B2E172d) on Wednesday 2020-06-16 12:09:29.

The withdrawal from Binance Platform was successful but never got confirmation in my Trezor Transaction History.

I can not to see the coins/tokens in my account (public address). I don’t know how to access the VET in my Public Address.
Please can someone help me with how to access the VET Tokens in my Trezor.

Any help would be much appreciated.

Thank You

matejcik, 2020-06-26:
> Hi there. I was wondering if anyone could help me please.
>
> I sent VeChain (VET) from Binance to the Trezor T Model (via MyEtherWallet) the Public Address: [...]

hello, this is a developer forum. Please contact Trezor Support through https://trezor.io/support/

commit bd5b3a3f (2020-05-29): common: drop Theta and VeChain ERC tokens, refresh market caps

fixes #1022
fixes #1023