trezor/trezor-firmware #1025: Not able to send transactions in RC3 with wallet or suite

bosomt, 2020-05-27:
**Describe the bug**
Not able to send transactions in RC3 with wallet or suite.
i see error + device freezes in status **Signing transaction**

**Firmware version or revision**
2.3.1 RC3

**Desktop/smartphone setup (please complete the following information):**
 - wallet + suite
 - nixos + fedora
 - chrome, app, firefox
 - latest

**To Reproduce**
Steps to reproduce the behavior:
try to send transaction

**Screenshots**
[image]

[image]

[image]

bosomt, 2020-05-27:
both BTC and ZEC works fine with 2.3.0

tsusanka, 2020-05-27:
What revision and URL of Suite and Wallet?

bosomt, 2020-05-27:
wallet.trezor.io
+
https://suite.corp.sldev.cz/suite-web/develop/

bosomt, 2020-05-27:
and latest appimage build from main branch from this morning

tsusanka, 2020-05-27:
Please reopen if this happens on the URLs mentioned in the testing table.

bosomt, 2020-05-29:
same error with latest T1 RC
Error details: Encountered invalid prevhash

tsusanka, 2020-05-29:
@sorooris you marked BTC sending as OK in the testing table, so you did not experience this? Could you try to reproduce @bosomt's bug?

sorooris, 2020-05-29:
on it

tsusanka, 2020-05-29:
@sorooris @bosomt could you please try out Litecoin as well?

wendys-cats, 2020-05-29:
@tsusanka LTC tx on 2.3.1 is OK, used wallet on : https://mytrezor.corp.sldev.cz/baking-crocodile/development

tsusanka, 2020-05-29:
Ok, so this seems to concern Zcash solely and is most likely a Wallet/Suite issue. I will contact the the Suite team.

bosomt, 2020-05-29:
LTC transaction via suite OK

fw: RC3: revision: 5209804
suite revision: 30cb3eb4261cca9376e0a0cd92aeba7470ccb984

1000101, 2020-05-29:
This could well be a problem with backends we are running. There have been some notable changes lately, we have bumped both backends to the same version a while ago (not latest, though). Could you please try it once again?

tsusanka, 2020-05-29:
@bosomt @wendys-cats could you please retry? But I do not think it was a backend issue. But just so we are sure.

sorooris, 2020-05-31:
Suite version: [develop + connect zcash sign fix](https://suite.corp.sldev.cz/suite-web/develop/accounts/send/#/zec/1?trezor-connect-src=https://connect.corp.sldev.cz/fix/zcash-sign/) 30cb3eb4261cca9376e0a0cd92aeba7470ccb984
Wallet version: [baking-crocodile](https://mytrezor.corp.sldev.cz/baking-crocodile/development/#/device/37B2926FE750EA9E1D7FF6AD/account/0/send) 97ec358a0bdfa382402e962b76ba0739533c3b16

I was able to send several ZEC transactions with the released T1 [1.8.3](https://zec1.trezor.io/tx/c72d51890ea05c3e2c574d2d3e65354e8451c050c5eeac008b09aefd30f2be99), [1.9.0](https://zec1.trezor.io/tx/a88605a7c901a845b83a55090bad97f32f44317de3cb292ed974ac5870725a02) and TT [2.3.0](https://zec1.trezor.io/tx/c20978b1a645c4c7cd15135b15059dca1343ab9c497ced0cd6d945c8b02e68e5) firmwares in both Suite and Wallet.

The ZEC issue still persists in both Suite and Wallet after retesting today, using 1.9.1 and 2.3.1 RCs.

Devices: 
**TT 2.3.1** RC3 revision **5209804f**
**T1 1.9.1** revisions **5209804f** and **81553df0**.

[image]
[image]
[image]

bosomt, 2020-06-01:
ZCASH transaction fails with Electrum too

FW version: 2.3.1 RC3 5209804f

{"name":"Error","message":"Encountered invalid prev_hash","stack":"Error: Encountered invalid prev_hash\n at e.exports._handleRequests (/nix/store/a206vf55r6bn59q36syl0ilsa7yw8iqh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:4479403)\n at c.concurrency (/nix/store/a206vf55r6bn59q36syl0ilsa7yw8iqh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:4477864)\n at runMicrotasks (<anonymous>)\n at processTicksAndRejections (internal/process/task_queues.js:89:5)\n at async e.exports._runSession (/nix/store/a206vf55r6bn59q36syl0ilsa7yw8iqh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:4477927)\n at async Object.e.exports [as signBitcoinTransaction] (/nix/store/a206vf55r6bn59q36syl0ilsa7yw8iqh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:205731)","code":"Failure_ProcessError"}

prusnak, 2020-06-01:
> ZCASH transaction fails with Electrum too

You meant Exodus, no?

bosomt, 2020-06-01:
@prusnak of course Exodus, i was thinking about Zcash Electrum but testing via Exodus

sorooris, 2020-06-02:
Suite - **QA OK**, several ZEC transactions were made
Suite version: [develop](https://suite.corp.sldev.cz/suite-web/develop/wallet/accounts/?trezor-connect-src=https://connect.corp.sldev.cz/fix/zcash-sign/#/zec/0) + [connect zcash fix](https://github.com/trezor/connect/pull/596/commits/6db36c5efb8fcd8d8699d7f29a6a8f5470a65e6d)

Devices:
T1 **1.9.1** revision **81553df0**: sent [tx](https://zec1.trezor.io/tx/7df02c0a140f76c9de2b8bc1195ad87709a5db56e1616de80b079b129dae7b55); receive - t1XoMddveMGpTgQjKtwjmQp49XMpUJjqTN4
TT **2.3.1** revision **5209804f**: sent [tx](https://zec1.trezor.io/tx/b9ef073b466124dceaf3f72e3002c5c6f178b7758b72e2bdd9e03650acebec9f); receive - t1SLYvzpFdwQeNKKvdG5kgCDihGXjrA3G71

Wallet will be tested as soon as a fix is available.

sorooris, 2020-06-03:
Wallet - **QA OK**, several ZEC transactions were made
Wallet version: staging 312d4034295dbde990da7417b129d9db237ddaa0

Devices:
T1 1.9.1 c6b2580cd _signed_: sent [tx](https://zec1.trezor.io/tx/0cc50dd06b2744caeb9ae68865b82da67351834b793a1f820aa26a37afce4f43); receive - 3K6bcA5mmq6BRVq6BQd1g2TngFrZF3x92g
TT 2.3.1 c6b2580cd _signed_: sent [tx](https://zec1.trezor.io/tx/c22fec279ab5c3bc1d1a6c0e5244cea2db207c31e40a716e247a537d6a566e88); receive - t1cVkbG3iKTBp8sYDxYDRfGdb5yWzoxN6TS

--**Ready to ship.**--

tsusanka, 2020-06-03:
Just for the record: this was a fronted issue and was fixed by https://github.com/satoshilabs/mytrezor/pull/928 and https://github.com/trezor/connect/pull/596.

closed 2020-06-03 by tsusanka