trezor/trezor-firmware #1026: trezorctl claiming "Failed to find device" when the device is in use by Suite

matejcik, 2020-05-27:
trezorctl coerces any errors while connecting to device into "Failed to find device".
Particularly in case of Bridge, we can detect that the device is acquired by a different user (by the presence of `session` field in enumeration) and show an appropriate response.

This also affects `trezorctl list`, which errors out in presence of an already acquired device.

Bonus points for implementing a command that force-releases the device. This could be merged with `trezorctl usb-reset` #225

tsusanka, 2020-06-04:
@szymonlesisz is this something high-priority for you?

closed 2022-06-30 by matejcik

STew790, 2022-08-01:
What can be done here by QA?

matejcik, 2022-08-01:
QA scenario:

1. Open Trezor Suite, go to Receive tab, show an address on Trezor.
2. **While the address is on screen** and Suite is waiting for user response, test the following:
3. run `trezorctl list`. You should not see a crash. Trezorctl should correctly report that device is in use.
4. try several different `trezorctl` commands. In each case, trezorctl should correctly respond that the device is in use.

STew790, 2022-08-02:
**QA NOK**
I tried all the listed commands on our wiki for trezorctl and here are the ones that show `Failed to find a Trezor device`
```
trezorctl get-address
trezorctl backup-device
trezorctl disable-passphrase
trezorctl reset-device
trezorctl set safety-checks
trezorctl get-features
trezorctl set-homescreen
```
**QA OK** 
All the other ones shows correctly `Device is in use by another process.`

The steps where the same as you mentioned.

matejcik, 2022-08-02:
Thanks for the report. That seems to be hitting a slightly different issue, so I'm going to move this one to approved.

commit 4b96e58c (2022-06-29): feat(python): detect busy device as a separate condition

fixes #1026
commit 8049282b (2022-06-30): feat(python): detect busy device as a separate condition

fixes #1026