trezor/trezor-firmware #1028: Electrum - error during signing a BTC transaction

sorooris, 2020-05-29:
Electrum 3.3.8
**Device**: TT 2.3.1 RC3, revision 5209804

**Steps to reproduce:**
1. navigate to the Send tab in Electrum
2. enter an arbitrary BTC address and amount; set any fee
3. confirm the Send button

**Expected result:**
Transaction is signed successfully.

**Actual result:**
Dialog with the following error is displayed: _b"U\x99\x11\xecJ\r%\xba\xe3\x91O\xc7$a\x15\xf3\xe1\xd3H%*\xda\x0b\x10\x9c\xb59(#|'\xf4"_ and the transaction is not signed; the device's _signing transaction_ animation freezes (see screenshot below) and can be fixed by any action in the wallet that needs to be displayed on the device.

**Screenshot(s):**
[image]

tsusanka, 2020-05-29:
@matejcik I guess this is because Electrum is not ready for the streaming right?

matejcik, 2020-05-30:
most likely correct

sorooris, 2020-05-31:
Electrum 3.3.8
Devices:
T1 1.8.3, revision df0963ec48f01f3d07ffca556e21ff0070cab099
T1 1.9.0, revision 0b7a8449f8dd003fc415262b05102d113247d3de
TT 2.3.0, revision 0b7a8449f

Retested the issue and I was able to sign and broadcast BTC transactions ([1](https://btc1.trezor.io/tx/8b45e47509b768f9ee91a74811910394d80503ec8e24ec220acff7075d575dd5), [2](https://btc1.trezor.io/tx/f391bef87ae4926caa794bd5aeadeedd4fb44e9a45f7f269b478437cbcfe0eae) and [3](https://btc1.trezor.io/tx/d3cc921f5889ff33e7bd34a65abd31726de6e624113df52466fe734fa41c3216)) from bech32 to segwit in Electrum using the **previously released versions of firmware**.

tsusanka, 2020-06-03:
@matejcik I guess I can close this in favour of https://github.com/spesmilo/electrum/pull/6198?

matejcik, 2020-06-04:
right.

closed 2020-06-04 by matejcik