trezor/trezor-firmware #1029: Allow only BIP69 transactions?

prusnak, 2020-06-01:
Do we want to allow only BIP69 transactions? (lexicographically sorted inputs and outputs). Our web wallet already does this, but what about the others?

- [ ] Trezor Wallet
- [ ] Trezor Suite
- [ ] Trezor Connect
- [ ] Bitcoin Core
- [ ] Electrum
- [ ] Wasabi

destrys, 2020-06-01:
Is the proposal here to have Trezor devices only able to sign BIP69 transactions?

If so, I think this isn't a good idea.
BIP69 isn't part of consensus, it's still just proposed.
I don't see a convincing reason to have a hardware wallet add a layer of restriction. 
It seems to me that BIP69 is a wallet spec, not a keystore spec.

Interestingly, this would introduce a tiny data leak, any non-BIP69 transaction would be known to be signed by a non-Trezor.

prusnak, 2020-06-01:
Yes, this was a meta-task to document the current state of things. I still think BIP69 is good, but we will not enforce it in Trezor for now.

closed 2020-06-01 by prusnak