trezor/trezor-firmware #1030: Support pre-overwinter inputs

tsusanka, 2020-06-01:
In 2.3.1 and 1.9.1 pre-overwinter inputs are not be spendable. Since overwinter has been activated in 2018 already maybe this does not pose a problem? We have decided we will wait ~two weeks if this pops up at support to see if this needs to be addressed.

tsusanka, 2020-06-10:
Same goes for prevtx with shielded outputs.

tsusanka, 2020-06-10:
Just for the record: slush encountered exactly this in #1059.

tsusanka, 2020-06-14:
One ticket on support. If this is easy let's fix it.

tsusanka, 2020-06-19:
Since we need Suite's cooperation here it would be nice to do this asap. @matejcik does this also mean we will introduce the `overwintered` field in SignTx, which we have removed in 2.3.0?

prusnak, 2020-06-19:
I don't think we need this, although it might simplify some things.

IIRC overwintered is equivalent to "coin is Zcash(-like)" and "version is 1 or 2".

tsusanka, 2020-06-19:
One way or another we have removed `overwintered` from Signtx and @szymonlesisz found that out just about now (we have to improve the fw <-> Connect relationship, see #1076) so we need to decide whether we have removed it forever or not.

matejcik, 2020-06-22:
@prusnak is correct, we can use `tx.version < 3` or something, also `version_group_id is not None`.
(in fact we should check that the fields match)

closed 2020-06-30 by matejcik

tsusanka, 2020-07-03:
Ok so I have finally tested this. I have used the [Connect Explorer](https://sisyfos.sldev.cz/connect-explorer/#/method/signTransaction-zcash) and with the same setup (seed, inputs, outputs):

1. on 2.1.8 I have successfully signed a transaction
1. on 2.3.1 I have encountered the "Invalid prev hash error"
1. on master (25a2e690fee0e79e581863a946b030c77e9717d2) I have successfully signed the transaction! The signed transaction was the same as in (1)!

Although I'd be happier if we would test this in Suite directly, I am fairly confident this works everywhere. So I am considerring this tested.

Details about how what data I have used:

<details>

```
[{
    address_n: [2147483692, 2147483781, 2147483648, 0, 0],
    prev_hash: '29d25589db4623d1a33c58745b8f95b131f49841c79dcd171847d0d7e9e2dc3a',
    prev_index: 0,
    amount: '80000',
}]

[{
    address: 't1N5zTV5PjJqRgSPmszHopk88Nc6mvMBSD7',
    amount: '72200',
    script_type: 'PAYTOADDRESS',
}]

version = 4
group id = 03c48270
overwintered = true

2.1.8:
{
	payload: {
		signatures: [
			'3044022076929db37a97d0e88dbf97a6edd9f6a5ff9cc61a0fc05339f9222d45249fa58002200fbbcee7f266aba62609154b48e3763a35b24e89c40ad10108b7abfce7327ec0'
		],
		serializedTx: '0400008003000000013adce2e9d7d0471817cd9dc74198f431b1958f5b74583ca3d12346db8955d229000000006a473044022076929db37a97d0e88dbf97a6edd9f6a5ff9cc61a0fc05339f9222d45249fa58002200fbbcee7f266aba62609154b48e3763a35b24e89c40ad10108b7abfce7327ec00121022f5c53b6d2e1b64c37d85716dbef318bd398ad7d2a03d94960af060402380658ffffffff01081a0100000000001976a9142e383c56fe3df202792e6f4460c8056b6a4d5b3488ac00000000000000000000000000000000000000'
	},
	id: 15,
	success: true
}

2.3.0:
{
	payload: {
		error: 'Encountered invalid prev_hash',
		code: 'Failure_ProcessError'
	},
	id: 12,
	success: false
}

master
{
	payload: {
		signatures: [
			'3044022076929db37a97d0e88dbf97a6edd9f6a5ff9cc61a0fc05339f9222d45249fa58002200fbbcee7f266aba62609154b48e3763a35b24e89c40ad10108b7abfce7327ec0'
		],
		serializedTx: '0400008003000000013adce2e9d7d0471817cd9dc74198f431b1958f5b74583ca3d12346db8955d229000000006a473044022076929db37a97d0e88dbf97a6edd9f6a5ff9cc61a0fc05339f9222d45249fa58002200fbbcee7f266aba62609154b48e3763a35b24e89c40ad10108b7abfce7327ec00121022f5c53b6d2e1b64c37d85716dbef318bd398ad7d2a03d94960af060402380658ffffffff01081a0100000000001976a9142e383c56fe3df202792e6f4460c8056b6a4d5b3488ac00000000000000000000000000000000000000'
	},
	id: 14,
	success: true
}
```

</details>

seperatis, 2020-08-06:
Updated to new firmware today. Still cannot send Zcash.

`Failed to send transaction
Error details: specified a negative value for writing an unsigned value`

prusnak, 2020-08-06:
@seperatis This is unfortunately different issue in a library we use. We've reported the issue here and we are investigating it too: https://github.com/BitGo/bitgo-utxo-lib/issues/83

PR #1079 (2020-06-30): Support pre-overwinter inputs
fixes #1030 

I use `tx.version >= 3` to detect whether to use overwinter logic.

legacy code properly checks that `version_group_id` is provided if and only if `version >= 3`. core code _does not check this_, because there doesn't seem to be a good place for this check: the place would be in `helpers.sanitize_tx_meta`, but "`version >= 3`" is too much business logic that I don't think belongs there.

IMHO the proper solution is to have a set of functions `BitcoinLike.sanitize_*`, that by default call `helpers.sanitize_*`, but can perform their own checks. I consi [... cut]
commit e6e30430 (2020-06-30): all: implement support for pre-overwinter prevtx (fixes #1030)