trezor/trezor-firmware #1032: Binance login with two devices

wendys-cats, 2020-06-02:
**Issue: Two devices used for login on binance.com**

Tested on two T1 devices (1.9.0 signed, 1.9.0, usigned, 1.9.1 FW) and with one TT device (2.3.1. FW)  - will most likely work with any other two devices. 
combinations: 1.9.0S and 1.9.0S, 1.9.1. and 1.9.0S, 1.9.1 and 1.9.0U, 1.9.1 and 1.9.1, 1.9.1 and 2.3.1
19.1. FW used trezor-fw-regular-1.9.1-81553df0 
2.3.1 FW used trezor-fw-regular-2.3.1-5209804f
Binance.com website
Chromium browser Version 81.0.4044.113 (Developer Build) (64-bit)
NixOS stable

**Bug Description**
If user logs in with another device than they have registered at first, Binance will allow the login (OK). But at the same time, this action of using another device "locks out" the first device - the one they have used initially for registrating a security key and when user tries to either log in with this first device, or delete the security key, it will result in unspecified 2FA verification code error.

**video for deleting key:** https://drive.google.com/file/d/1BNTmqiwUjg8rJG21uxE3dLFfATuzcxHa/view
**video for login:** https://drive.google.com/file/d/1_HVxVrwbTqfzvvtuxF_w_-YXYnV9Yx7F/view

**Steps to Reproduce**
**1.** Two devices with same seed
**2.** Register device A on Binance as a security key
**3.** Login with device B to Binance
**4.** Try to login or delete security key with device A

**Expected behavior**
User should be able to swap devices as needed on Binance, as long as they have the same seed.

matejcik, 2020-06-02:
> **Expected behavior**
> User should be able to swap devices as needed on Binance, as long as they have the same seed.

that is very much not true in general.
did this work with previous firmwares?

prusnak, 2020-06-02:
In general, that is not true. However, Binance does not use the signature counter - so it should work.

wendys-cats, 2020-06-02:
Binance does not have counter like GitHub does. The case and the expected scenario both specify Binance, not "general", so the expected behavior stands true. 
Not sure if it worked with previous firmwares, @sorooris @bosomt ?

bosomt, 2020-06-02:
yes it's broken

tetsed with two devices with same seed

bosomt, 2020-06-02:
maybe it's problem that

T1 uses U2F 
and TT is used as FIDO2 device,
so maybe binance will disable/expects  U2F device next time

tsusanka, 2020-06-02:
What I am now mostly interested in now is whether this is a regression or not. @bosomt @wendys-cats could you please test with 1.9.0 and 2.3.0 for example?

andrewkozlik, 2020-06-02:
Firstly:
- U2F always uses a signature counter.
- FIDO2 uses a signature counter unless we explicitly know that the site is OK with us not using a signature counter.
- T1 supports U2F only.
- TT supports both U2F and FIDO2, so it depends on the browser which type of credential is created. I think Firefox still does only U2F, while Chrome does FIDO2.

The problem is this:
- If you register with a T1, then clearly you aren't using FIDO2, but U2F, and that always uses a signature counter, so you can't switch between devices.
- If you register with a TT and a browser which doesn't support FIDO2, then the same applies.
- If you register with a TT and a browser which does support FIDO2, then you won't be able to log in using a T1 even if it has the same seed, because it won't recognize the FIDO2 credential.

andrewkozlik, 2020-06-02:
So just to summarize, there is no way that you can use a T1 and a TT to alternately log in to a site.

andrewkozlik, 2020-06-02:
**Expected behavior**
User should be able to swap **_Trezor T_** devices as needed on Binance, as long as they have the same seed _**and the browser used supports FIDO2**_.

wendys-cats, 2020-06-02:
@andrewkozlik thank you for a more detailed info about how FIDO and U2F work:)

This here - 
"If you register with a T1, then clearly you aren't using FIDO2, but U2F, and that always uses a signature counter, so you can't switch between devices."
 - in this case, there is another problem, because I did switch devices. I registered device A, then logged in with device B (see in Steps to reproduce).

andrewkozlik, 2020-06-02:
> This here -
> "If you register with a T1, then clearly you aren't using FIDO2, but U2F, and that always uses a signature counter, so you can't switch between devices."
>
>     * in this case, there is [...]

I see, well to be more precise, you have no guarantee of being able to switch between devices. It depends on their counter values. The relying party keeps track of the last counter value used and checks that it's increased for every operation. It doesn't have to be by one, it can be more. So for example, if device A has its counter at 50 and B has it at 55, then something like this might happen:
1. Register with A and a counter value of 50. Success.
2. Authenticate with A and a counter value of 51. Success.
3. Authenticate with B and a counter value of 55. Success.
4. Authenticate with A and a counter value of 52. Fail.
5. Authenticate with A and a counter value of 53. Fail.
6. Authenticate with A and a counter value of 54. Fail.
7. Authenticate with A and a counter value of 55. Fail.
8. Authenticate with A and a counter value of 56. Success.

Some relying parties might ban the credential at step 4.

tsusanka, 2020-06-02:
So to summarize, switching between device is not supported, it _may_ work in some cases, and it _must_ always work for the cases mentioned [above by Andrew](https://github.com/trezor/trezor-firmware/issues/1032#issuecomment-637459335). QA should update their test scenario accordingly, only the behavior described by Andrew makes sense to test.

Correct? I am closing then.

closed 2020-06-02 by tsusanka