trezor/trezor-firmware #1038: Segwit prev_tx streaming is quite slow on TT

tsusanka, 2020-06-04:
I have measured how long signing takes for 100 inputs:
- On T1 without prev_tx streaming: 25s
- On T1 with prev_tx streaming: 1m 10s.
- On TT without prev_tx streaming: 50s
- On TT with prev_tx streaming: **10m 35s**.

Logs of communication:
[t1-no-streaming.txt](https://github.com/trezor/trezor-firmware/files/4730194/t1-no-streaming.txt)
[t1-streaming.txt](https://github.com/trezor/trezor-firmware/files/4730195/t1-streaming.txt)
[tt-no-streaming.txt](https://github.com/trezor/trezor-firmware/files/4730196/tt-no-streaming.txt)
[tt-streaming.txt](https://github.com/trezor/trezor-firmware/files/4730197/tt-streaming.txt)

The protocol seems to be the same in both scenario.

The test is in present in `tsusanka/streaming-timing`, I'll be glad if someone reviews my measurement just to be sure.

The first course of action should be to find out where is the bottleneck (protobuf is our main contender).

prusnak, 2020-06-06:
There is this older PR by Honza reworking the Protobuf internals: https://github.com/trezor/trezor-firmware/pull/190 If we decide to address this issue, we might want to revive that PR.

hiviah, 2020-06-09:
For the record, measurement with GC calls from `sign_tx` removed (`gc.collect` calls):

* 59 G instructions in emulator with GC calls removed, 359 G instructions with explicit GC calls (so about 6x faster with default 20M heap)
* 9:13 min on device with GC calls removed, so about 14% faster than the original measurement.

Apparently the new protobuf calls cause lot of allocations and load in garbage collector calls, which can be seen on device when sampling by ITM/ETM trace.

matejcik, 2020-07-13:
partial fix: #1089 

another step is removing explicit `gc.collect` in helpers, per @hiviah's suggestion

even with all this, protobuf is relatively slow and we'll need to look for more

tsusanka, 2020-07-24:
Closed by #1089 and #1128.

closed 2020-07-24 by tsusanka

tsusanka, 2020-08-05:
QA has tested this as part of the freeze testing.