trezor/trezor-firmware #1039: Remove Connect to host dialog in bootloader

tsusanka, 2020-06-04:
Something to consider. It might seem redundant at the moment. If you wish not to communicate with computer why are you plugging the Trezor into the computer in the first place?

A counter-argument might be that in firmware I am protected using PIN (only after that the USB is turned on) and this servers as a confirmation I want to indeed enter the bootloader - I might have entered by accident.

tsusanka, 2020-06-05:
> A counter-argument might be that in firmware I am protected using PIN (only after that the USB is turned on) and this serves as a confirmation I want to indeed enter the bootloader - I might have done it by accident.

We have decided this argument is not strong enough and we want to go this way.

tsusanka, 2020-06-05:
Also note that while removing this dialog we will remove the option to display the firmware fingerprint:

[image]

Let's move this feature (clicking on the (i) icon) into the second screen - the one that will be directly loaded without confirmation.

mcudev, 2021-01-20:
This is a security feature. It's there as a conservative protection in-case there is an exploitable bug in the USB stack. The USB stack is pretty complex. Maybe after a couple of years in production now there is more confidence in it, and this is less necessary. I think that the vision was to have more on-device management capabilities available in the bootloader. For example, not starting the USB stack is nice if you have a device that is setup and you want to wipe it with the on-device management capabilities. In that case, you don't need the extra risk that a USB bug could theoretically impose. Not starting-up USB easily avoids a chunk of attack surface. I always thought this was a nice extra step of protection.

If this protection is disabled, I suppose a very diligent user could wipe their device in an offline fashion by putting a bootloader on a microSD and letting the boardloader erase the flash.

tsusanka, 2021-02-19:
Will be done as part of https://github.com/trezor/trezor-firmware/issues/1049.

closed 2021-02-19 by tsusanka