trezor/trezor-firmware #1045: Modify T1 bootloader to run both Core and Legacy

tsusanka, 2020-06-05:
As part of #24 we want to run Core on T1. We need to modify the T1 bootloader to support this. As a product requirement we also want to support Legacy (up from version 1.9.x is enough).

prusnak, 2020-06-05:
I think this is a no-op. Since bootloader 1.8.0 we changed the firmware header to match the firmware header of core firmware, so I think it SHOULD be possible to flash the SIGNED core firmware from master built for T1 on device with BL 1.8.0. We'd need to test this, though, so please let's keep this open.

tsusanka, 2020-12-18:
I think we can close this one? Becuase Core is running on T1 right? cc @mmilata @hiviah

hiviah, 2020-12-18:
Yes it can be closed. I think there is only one detail that you can't run unsigned core because T1 production bootloader won't let you change interrupt vector table.

tsusanka, 2020-12-18:
And that's why we have the Trezors you provided as right?

hiviah, 2020-12-18:
Yes, they have bootloader with `MEMORY_PROTECT=0`.

tsusanka, 2020-12-18:
Awesome, thanks :rocket:.

closed 2020-12-18 by tsusanka