trezor/trezor-firmware #1047: Jump from firmware to bootloader without restarting

tsusanka, 2020-06-05:
- From some on-device menu for example.
- ButtonRequests are needed when this happens.

@slush: We only need to evaluate security implications. I think special button confirmation before jumping to bootloader should be enough though.

slush0, 2020-06-05:
To prevent misunderstanding, the requirement is that the jump is initiated by host over API, and confirmed by the user on device.

tsusanka, 2020-06-05:
Ah, so do we want to go the way you are mentioning or using the on-device menu? Or both?

I haven't realized your way makes way more sense - it is initiated by the host and we do not need to explain the user how to enter the bootloader.

slush0, 2020-06-05:
Exactly this

hiviah, 2020-06-09:
Not sure if this is possible. Normally firmware drops privileges at start, though through SVC call it might be able to change the VTOR table back and jump to bootloader.

tsusanka, 2020-12-21:
@prusnak @hiviah could we assess if this is possible in any way? @matejzak was interested in this a lot since it would simplify the onboarding in Suite (they have to teach users how to enter bootloader etc.).

prusnak, 2020-12-21:
It is possible, but needs change to the bootloader.

We can soft-reboot the device, but just before doing that we'd store a flag in some register that is not cleaned during the soft-reboot (IIRC there are some). When the device boots itself, the bootloader will check this special flag to determine whether it should stay or whether it should start the firmware.

tsusanka, 2021-01-14:
We would like to do this for both T1 and TT bootloader.

mcudev, 2021-01-20:
@prusnak Unless we're thinking of different mechanisms, I think a soft reset will take us back through the boardloader and the first thing it does inside of `main` is to check for a non-power-on reset event (including soft resets) and shutdown. I think one reason why that's in the boardloader is to mitigate some glitching attacks that do not attack the MCU boot ROM.

When I first read this, my gut reaction was, "that's dangerous". After pondering, it still seems inadvisable, but it might be acceptable, **if** it's possible to get the privileges back and jump back to the bootloader. I think we'd have to account for the possibilities that 1) there is sensitive data left by the firmware somewhere and 2) the integrity and authenticity of the firmware that we are jumping to is no longer valid. The first would be covered by using the existing `jump_to` style functions that clear stuff out. The second would be covered by doing the same signature check of the bootloader that the boardloader does. Also, when jumping backwards in the process, it seems like we could run into peripheral re-initialization bugs, some non-trivial amount of testing might be necessary. Technically, I think it's best to just unplug and re-plug the device. That way, the bootloader will be run in it's most tested state, it's just a matter of staying in the bootloader. If rubbing the screen is too complicated, then what else could we do?

tsusanka, 2021-01-29:
Note that this is kind of blocker for https://github.com/trezor/trezor-suite/issues/2338. We can discuss on standup if it really depends on this but IMHO it does.

andrewkozlik, 2021-02-16:
For the record, here is an idea on how we could use the flash to solve this problem if we can't find a solution which uses volatile memory. We have two flash sectors that we use for the norcow storage. One begins with 0x3243524e and contains the stored data. We call this the "active" one. The other sector is filled with 0xff bytes. The bootloader could do this:
1. Find the first norcow sector that starts with 0xffffffff. (If not found continue to firmware.)
2. Starting at offset 4 in this sector find the first non-zero byte. (If not found continue to firmware.)
3. If the byte is 0xff then go to firmware, otherwise set the byte to 0x00 and continue to bootloader.

When the firmware wants to restart in bootloader mode, it would find the first non-zero byte as in steps 1 and 2 above and set it to 0x55.

To explain some details:
- We can change any bit in the NOR flash from 1 to 0 easily. So 0xff -> 0x55 -> 0x00 are cheap operations, whereas to change a bit from 0 to 1 you need to reset the entire sector to 0xff. I think resetting the sector degrades the flash, so we don't want to do it more than we have to.
- The reason why we have two sectors is that every once in a while we need to compactify the sector and we do that by copying the data from one sector to the other, because we can't keep the whole thing in RAM. So in exceptional circumstances both sectors could have data written to them, like if there is a power outage during compactification, but that's not a problem in this case, because in that case the next boot will go to the firmware as expected.

tsusanka, 2021-03-04:
Done by T1 for now. @prusnak will revisit TT possibilities.

tsusanka, 2021-04-15:
On a product meeting we have decided:

- We will modify the boardloader to allow this feature on TT. @prusnak will take care of that.
- We will not do https://github.com/trezor/trezor-firmware/issues/1047#issuecomment-780026056. We would have to deal with yet another option in the onboarding design and it seems not worth it at the moment.
- Dialog confirmation is required before jumping to bootloader.

tsusanka, 2021-04-15:
To clean this up a bit, I am closing this issue after #1570 is merged and I'll create a new one for TT.

closed 2021-04-15 by prusnak