trezor/trezor-firmware #1050: Transaction on Ethereum based network with different chainID not able to be signed

cryptomar1o, 2020-06-05:
It is not possible to sign transactions on some Ethereum based networks. Common for these networks is that they look like an Ethereum network but using different ChainId. Instead of signing a transaction, the error message _Forbidden key path_ is shown by the used application.

Changing the derivation path makes no difference.

The issue seems to be connected to the firmware version 2.3.1 only. Ethereum Kovan tested with firmware version 2.3.0 and works. Also firmware 1.9.1 signs the transaction on Ethereum Kovan with no problems.

**Examples:**
Ethereum Kovan - possible to access through MyEtherWallet, MyCrypto, MetaMask (faucet for tokens: https://faucet.kovan.network/)

Volta - possible to access through MyCrypto (faucet for tokens: https://voltafaucet.energyweb.org/)

EWC - possible to access through MyCrypto

**Networks with no problem (transaction is signed):**
Ethereum Ropsten
Ethereum Classic

**Error messages:**
MyCrypto
Does not communicate at all with Trezor device

MetaMask
> Error: Forbidden key path

MyEtherWallet:
> {
  "exception": {
    "values": [
      {
        "type": "Error",
        "value": "Forbidden key path",
        "stacktrace": {
          "frames": [
            {
              "colno": 62077,
              "filename": "https://www.myetherwallet.com/js/vendors.890b46c4.js",
              "function": "s",
              "in_app": true,
              "lineno": 797
            },
            {
              "colno": 61866,
              "filename": "https://www.myetherwallet.com/js/vendors.890b46c4.js",
              "function": "r",
              "in_app": true,
              "lineno": 797
            },
            {
              "colno": 30439,
              "filename": "https://www.myetherwallet.com/js/vendors.890b46c4.js",
              "function": "Generator.forEach.e.<computed> [as next]",
              "in_app": true,
              "lineno": 851
            },
            {
              "colno": 31354,
              "filename": "https://www.myetherwallet.com/js/vendors.890b46c4.js",
              "function": "Generator._invoke",
              "in_app": true,
              "lineno": 851
            },
            {
              "colno": 30012,
              "filename": "https://www.myetherwallet.com/js/vendors.890b46c4.js",
              "function": "l",
              "in_app": true,
              "lineno": 851
            },
            {
              "colno": 1501895,
              "filename": "https://www.myetherwallet.com/js/app.5813b6de.js",
              "function": "?",
              "in_app": true,
              "lineno": 1
            }
          ]
        },
        "mechanism": {
          "handled": true,
          "type": "generic"
        }
      }
    ]
  },
  "level": "error",
  "event_id": "6dfe49f2497642bba13cf8b1a342ddec",
  "platform": "javascript",
  "sdk": {
    "name": "sentry.javascript.browser",
    "packages": [
      {
        "name": "npm:@sentry/browser",
        "version": "5.16.0"
      }
    ],
    "version": "5.16.0",
    "integrations": [
      "InboundFilters",
      "FunctionToString",
      "TryCatch",
      "Breadcrumbs",
      "GlobalHandlers",
      "LinkedErrors",
      "UserAgent",
      "Vue"
    ]
  },
  "timestamp": 1591361427.6282263,
  "environment": "web",
  "release": "5.6.4",
  "request": {
    "url": "https://www.myetherwallet.com/interface/send-transaction",
    "headers": {
      "User-Agent": "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.75 Safari/537.36"
    }
  },
  "tags": {
    "network": "KOV",
    "service": "myetherwallet.com",
    "walletType": "trezor"
  }
}

prusnak, 2020-06-05:
Kovan is a testnet for Ethereum so it should use the `m/44'/1'` path

matejcik, 2020-06-06:
we have explicit support for Kovan: https://github.com/trezor/trezor-firmware/blob/master/common/defs/ethereum/networks.json#L108-L117
with chain_id 42 and `m/44'/1'/x'` it should work fine

We don't claim support for Volta nor EWC (not listed on trezor.io/coins), so I'd say that not supporting them is correct, what do you think @prusnak ?
If their users are interested, let them submit a PR to [`common/defs/ethereum/networks.json`](https://github.com/trezor/trezor-firmware/blob/master/common/defs/ethereum/networks.json)

prusnak, 2020-06-06:
> We don't claim support for Volta nor EWC (not listed on trezor.io/coins), so I'd say that not supporting them is correct,

Agree

matejcik, 2020-06-08:
Since version 2.3.1 it is expected behavior that mismatched chain_id and derivation path will fail to sign.

closed 2020-06-08 by matejcik