trezor/trezor-firmware #1051: Investigate coverage drop

prusnak, 2020-06-07:
It seems the coverage (in the `core unix coverage posttest` CI target) has dropped to 17%. My guess is something got broken in the coverage itself.

Let's find out what exactly it was and maybe introduce a rule which fails the coverage target in the CI if the percentage is decreased (or below a fixed value).

tsusanka, 2020-07-31:
@mmilata this is something you might look at in case you find yourself eager to do some github issue :). Might be good to grasp another parts of the code.

mmilata, 2020-08-13:
The coverage records contain out-of-range line numbers. This is caused by `mpy-tool.py` corrupting the line number information during module freezing. Opened upstream PR to fix it:  micropython/micropython#6335

tsusanka, 2020-08-16:
Btw we are using our forked micropython. So @prusnak should @mmilata file the PR into trezor/micropython as well and we merge it there?

prusnak, 2020-08-17:
Yes, @mmilata send PR against https://github.com/trezor/micropython - `trezor-v1.12` branch. Once merged here, we'll update the https://github.com/trezor/trezor-firmware/pull/1185 to use this updated branch.

PR #1185 (2020-08-18): Fix test coverage reports
Fixes #1051. 

Opening as draft until micropython/micropython#6335 is merged.

The PR also adds configurable coverage threshold which makes the CI fail if the coverage drops below. There is however a problem with the current implementation because when such failure happens, no artifact is generated and the developer won't be able to access the report. Not sure if GitLab CI allows jobs to produce artifacts even when they fail, or whether we need a separate job for this check.
closed 2020-08-18 by tsusanka