trezor/trezor-firmware #1059: Zcash signing fails on "Encountered invalid prevhash"

slush0, 2020-06-10:
**Describe the bug**
When sending transaction from Trezor to transparent Zcash address, signing fails on "Encountered invalid prevhash" message. Fails on old web wallet as well as on beta Suite (2020.05.01).

I was trying to spend input from tx ec1371e482fdd08836bf143b79b015184784448e93c626288d18563e8c66c7a2 and send it to transparent address, so no shielded inputs/outputs were involved.

**Firmware version and revision**
FW 1.9.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trzor Wallet, Trezor Suite Beta
 - OS: Linux
 - Browser: Firefox
 - Version 76

tsusanka, 2020-06-10:
@trezor/qa could you retest this please?

bosomt, 2020-06-10:
**FW** 1.9.1 production
**Wallet:** Trezor wallet  production
**Browser** Version 83.0.4103.61 (Developer Build) (64-bit)

I was not able to reproduce it.

From Transparent to Transparent address.
Worked fine.
Tx: https://zec1.trezor.io/tx/a2fbc27cbd68bd3f6c2111facf7028fae83470480ba5cf0874762b9fa269170f

[image]

tsusanka, 2020-06-10:
Ah I think I know! I believe @slush0 is trying to spend tx from pre-2018 (it seems it is [from 2017](https://explorer.zcha.in/transactions/ec1371e482fdd08836bf143b79b015184784448e93c626288d18563e8c66c7a2)). So this is exactly #1030.

slush0, 2020-06-10:
Ah, so there we go.

closed 2020-06-10 by slush0