trezor/trezor-firmware #1064: Allow third-parties BIP-32 paths and provide a setting

tsusanka, 2020-06-11:
As a follow up of #1044 we have decided we will:

- [ ] Allow [greenaddress](https://github.com/trezor/trezor-firmware/issues/1044#issuecomment-641186469) and [Casa](https://github.com/trezor/trezor-firmware/issues/1044#issuecomment-639594974)'s paths.
- [ ] Provide a new setting which disables the paths checks and allows all paths. This setting can be set using trezorctl and strongly informs the user that this is not a good idea and that it is possible it will not work in future. The paths above will work also without this setting.
- [ ] Strengthen the soft checks in get_address and similar to a hard fail unless the setting above is set.

This will strengthen our paths protections and make it clear using non-standard paths is unusual and may not be supported in future.

jmacato, 2020-06-22:
IDK if this is the correct place to report this but I'm currently trying to fix the incompatibility in 2.3.1 between Wasabi and Trezor devices.. Now the fixes works good on Mainnet but i kept failing and got invalid keypath exception while signing a transaction for Testnet. The keypath that failed is `m/84'/0'/0'/0/4`.

jmacato, 2020-06-22:
IDK if this is the correct place to report this but I'm currently trying to fix the incompatibility in 2.3.1 between Wasabi and Trezor devices.. Now the fixes works good on Mainnet but i kept failing and got invalid keypath exception while signing a transaction for Testnet. The keypath that failed is m/84'/0'/0'/0/0 and this prompt is shown twice...

[image]

prusnak, 2020-06-22:
The correct Testnet path is  `m/84'/0'/1'/0/0` (second element is the coin).

jmacato, 2020-06-22:
@prusnak i see... we are using the same keypath for all bitcoin mainnet/testnet/regtestnet. I wonder if that merits some exception?

prusnak, 2020-06-22:
No, it does not. I don't see any reason why you should use the same keypath for mainnet and testnet. Do you?

jmacato, 2020-06-22:
Yeah.. Sadly that's how it is on wasabi for now. It'll need some effort in refactoring the wallet to use different key paths for each nets. Guess we'll just have to not use testnet with trezor for now.

molnard, 2020-06-22:
You can load the same wallet on every network with Wasabi. With this change, the TestNet/RegTest wallets created with Trezor will reject signing. Is there a way to solve this in a way that maintains compatibility?

prusnak, 2020-06-22:
( moved the off-topic discussion to email )

tsusanka, 2020-06-30:
> Provide a new setting which disables the paths checks and allows all paths. 

Let's make the setting not as "disable hard-core checks", but in more general for example "enable power user mode" or better "disable security checks" or smth like that. We may use it for other stuff in future.

matejcik, 2020-06-30:
"Disable security checks" sounds reasonable, and also indicates what general sort of thing falls under this.

destrys, 2020-06-30:
I'm not a fan of the trezorctl option for disabling restrictions introduced by SatoshiLabs. Most users will not be comfortable with this option. If there are other hardware wallets without non-consensus restrictions, that will be a safer path than advising customers to use trezorctl.

I would also again like to highlight that there is not an accepted standard to reference when talking about "non-standard paths".
It's not clear to me what exploit you are trying to protect against with these restrictions. I am not aware of funds/secrets being stolen from Trezor protected keys using a 'non-standard path' attack.

tsusanka, 2020-07-24:
>  Strengthen the soft-checks in get_address and similar to a hard fail unless the setting above is set. 

Moved to #1137.

---

So I believe we can close this one @matejcik?

matejcik, 2020-07-24:
i think so too. closing.

closed 2020-07-24 by matejcik