trezor/trezor-firmware #1065: Wipe before reset and recovery

tsusanka, 2020-06-11:
As a follow-up of #1063 and #1056 we have agreed with @matejcik it would be nice to wipe the storage before reset/recovery to make sure the storage is in an expected state. Although the device should not receive any settings after #1063 this seems as a good precaution.

Not to confuse the Host we want to keep the device id unchanged.

1. We will introduce a new function which will be called before reset/recovery:

```
reset_storage():  # name to be determined
- remember device_id if any
- wipe
- storage init
- write the previous device id
- write storage version
```

2. We will add a new public field `initialized` which will be used in Features.

tsusanka, 2020-06-16:
Closed by #1072.

tsusanka, 2020-06-16:
QA: just a quick test against wallet.trezor.io and suite (do not test autolock though) please
- simple reset and backup and get an address
- simple recovery and get an address

Nothing should break, so this is a regression test.

closed 2020-06-16 by tsusanka

bosomt, 2020-07-21:
@tsusanka  in last month we wiped multiple devices and recovered single or shamir or super shamir seed without any kind of problem. /with latest master of 2.3.2 and 1.9.2 /

Is there something particular that we should check or is this enough for now ?

tsusanka, 2020-07-21:
Mmm I am bit confused here, you have filed https://github.com/satoshilabs/mytrezor/issues/932 which claims otherwise, no?

bosomt, 2020-07-22:
@tsusanka sorry for confusion, I still see **Recovery failed. Error details: Firmware error** in old wallet

wendys-cats, 2020-07-23:
This issue has been tested via Asana request from @tsusanka. It has been solved on 2.3.2, 1.9.x has not been affected with the bug. Full run here.
Tested on 2.3.1 (c6b2580cd), 2.3.2 (33d2bf4), 1.9.1 (c6b2580cd), 1.9.2 (33d2bf41) here and here. Comment per request from @tsusanka.

https://github.com/trezor/trezor-suite/issues/2124#issuecomment-662960496