trezor/trezor-firmware #1066: Unify pin_chached

tsusanka, 2020-06-12:
`pin_cached` differs now between T1 and TT. In Legacy ti is 

```
resp->pin_cached = session_isUnlocked() && config_hasPin();
```

and in T it is simply:

```
pin_chaced = config.is_unlocked()
```

- Should we unify it?
- Can we rename it to `unlocked`?

@prusnak any takes on this?

tsusanka, 2020-06-12:
@szymonlesisz are you using `pin_cached` in Connect? Any takes on this?

tsusanka, 2020-06-19:
Let's not close this before talked to the Suite team.

tsusanka, 2020-06-29:
This _should_ be fine for Suite.

QA:
Please test T1 on current master against https://suite.corp.sldev.cz/suite-web/feat/232-pinentry/wallet/ and wallet.trezor.io.

1. Enable PIN.
2. Enable autolock (set to 10s).
3. Wait for the lock.
4. Unlock.
5. Disable PIN.

Make sure the Wallet/Suite reacts to these events as it should. The gist of the test should be "make sure nothing has changed" which is quite hard to transform into scenario, sorry about that :-/. So basically anything that comes to your mind.

sorooris, 2020-06-30:
**QA NOK**
Auto-lock **time** is displayed _nowhere_.

device: T1
Suite: web branch [232-pinentry](https://suite.corp.sldev.cz/suite-web/feat/232-pinentry/wallet/)
trezorctl: 0.12

sorooris, 2020-06-30:
**QA NOK**
Auto-lock **time** is displayed _nowhere_.

device: T1
Suite: web branch [232-pinentry](https://suite.corp.sldev.cz/suite-web/feat/232-pinentry/wallet/)
trezorctl: 0.12

User has no way of knowing how long the duration before auto-lock is on their device:
- nothing during auto-lock setup (on-screen confirmation),
- trezorctl get-features yields nothing,
- nowhere in Suite.

tsusanka, 2020-06-30:
@sorooris the introduction of this setting into suite is a separate issue here: https://github.com/trezor/trezor-suite/issues/1946

---

> nothing during auto-lock setup (on-screen confirmation),

What do you mean by this? When you are changing auto-lock delay you are confirming it on the device. It is true though we have enabled the auto-lock by default for everyone (10 min). This was agreed upon with Product.

matejcik, 2020-06-30:
> * trezorctl get-features yields nothing,
> * nowhere in Suite.

This is identical to T1 where (again) auto-lock has existed for a long time.

We might want to revisit this with Product though?

tsusanka, 2020-06-30:
I do not see any rational to publish the value in features (remember we do not want to bloat them). And in Suite it will displayed in the settings... It seems enough to me.

----

@sorooris so unless you have found something else should we close again?

matejcik, 2020-06-30:
> I do not see any rational to publish the value in features (remember we do not want to bloat them) unless we have some use case for that. And in Suite it will displayed in the settings... It seems enough to me.

can't be displayed in Suite if we don't add it to Features. The device currently does not report the current auto-lock delay in any way.

tsusanka, 2020-06-30:
:facepalm: correct. I still think we should introduce something like GetSettings instead of putting this into Features. So I am suggesting to move this into a separate issue. Imho it is not related to this issue in particular.

sorooris, 2020-06-30:
@tsusanka Thank you for pointing out the separate issue, i wasn't aware of it.

Comparison of model T (2.3.2 24f803e9) and T1 (1.9.2 24f803e9) dialogue (both after entering the same command `trezorctl set auto-lock-delay 10`):

 T1|TT
 -|-
[image]|[image]

Model T asks the user to confirm the time on-screen, T1 does not.

tsusanka, 2020-06-30:
I do not think this is a problem per se (we will unify everything with the _port_) but let's reevaluate in Product. Still it is imho not related to this issue. Could you please open a new one for this (your latest comment)? And I'll open another one for the features.

matejcik, 2020-06-30:
Correct. This is not related to the current issue, we didn't modify user-facing behavior of T1 so this is not a regression.

wendys-cats, 2020-06-30:
@tsusanka , question about FIDO/U2F (stick said you will know the exact implementation)

I have a T1 1.9.2 24f803e with auto-lock. Device is soft-locked. When I want to sign into Dropbox, the request for registration will appear on T1 without unlocking it (no PIN required). Is this correct behaviour? 

I distantly remember something about auto-lock colliding with FIDO/U2F requests and the decisiion whether or not it should be removed in this particular case, but I don't know the conclusion.

tsusanka, 2020-06-30:
> I have a T1 1.9.2 24f803e with auto-lock. Device is soft-locked. When I want to sign into Dropbox, the request for registration will appear on T1 without unlocking it (no PIN required). Is this correct behaviour?

Yes, this is correct. There is no way to enter the PIN on T1 when doing U2F/FIDO (we can't somehow interfere the protocol and display our PIN matrix or whatnot).

tsusanka, 2020-07-07:
Since this is not an auto-lock issue I believe this is done. We have concluded the QA remarks will be evaluated in https://github.com/trezor/trezor-suite/issues/1946, https://github.com/trezor/trezor-firmware/issues/1088 and https://github.com/trezor/trezor-firmware/issues/1097.

closed 2020-07-07 by tsusanka