trezor/trezor-firmware #1067: QR code dialog buttons not working properly

onvej-sl, 2020-06-12:
Neither of the buttons on the QR code dialog is responding properly. I have to tap it multiple times to invoke an action.

Firmware 2.3.1. It affects only device (not emulator).

[image]

tsusanka, 2020-06-12:
@trezor/qa could you retest please?

bosomt, 2020-06-12:
@tsusanka tested, this is happening with every version of FW for a long time.

Tested it now with 2.3.1 and it's sometime happens and sometime now.

I can record video when i press QR/Address button or OK button 10 times and it won't happen.
And sometimes it won't respond 10 times in row. 

i can show you at Monday in office.

sorooris, 2020-06-15:
@tsusanka 
**QA NOK**
Wallet: prod, commit hash 312d4034295dbde990da7417b129d9db237ddaa0
Browser: any (tested Chromium)

Device fw versions:
- 2.3.1 c6b2580cd
- 2.3.0 0b7a8449f
- 2.1.8 8eb6ce08

Note: Confirming issue with both buttons - however, there is an inconsistent occurence of the described issue as @bosomt describes above.

bosomt, 2020-06-16:
i just tried few older production firmwares
result: 

2.0.8 OK
2.1.4 NOK

if you want I can try to find exact firmware 
when this bad response of QR/address button was introduced

andrewkozlik, 2020-06-25:
The bug is caused by a missing repaint check in `Qr::on_render()`, so the device keeps redrawing the QR code. It was effectively introduced in 36534325f09b17c4581f0eeb6210fc0d6982cdfe, although the distal cause was probably the fact that the function was already missing `self.tainted = True` before that.

There are several other `on_render()` methods where the repaint check is missing. Most of them have some sort of animation like `ui.display.loader()` or pulsating arrows, so in those cases it's intentional. In the Homescreen and Lockscreen classes the repaint check is handled in the dispatch method. So `Qr::on_render()` seems to be the only problematic case.

closed 2020-06-25 by andrewkozlik

bosomt, 2020-06-26:
QA OK
now I have 100% success when changing QR/address via touchscreen button

tested with current master f382f77c 2.3.2 
and current wallet Git Commit: 1f09c65d25eaf01aa3da804dfedf765b588d3559