trezor/trezor-firmware #1070: Upgrade tests do not work on NixOS

tsusanka, 2020-06-12:
```
$ ./download_emulators.sh
...
Shell for /home/tsusanka/.local/share/virtualenvs/firmware-SJjfUTvR already activated.
No action taken to avoid nested environments.
$ cd emulators/
$  ls ./trezor-emu-legacy-v1.6.2
./trezor-emu-legacy-v1.6.2
$ ./trezor-emu-legacy-v1.6.2
Failed to execute process './trezor-emu-legacy-v1.6.2'. Reason:
The file './trezor-emu-legacy-v1.6.2' does not exist or could not be executed.
```

What seems peculiar to me is the part about virtualenv. Isn't somehow accidentely the root shell.nix used? Weird..

prusnak, 2020-06-18:
The script works as expected if I run it from `trezor-firmware/tests` without nix-shell (and thus virtualenv activated).

If I am in activated nix-shell (and thus virtualenv) I indeed see the same error (and the files are not patched via autopatchelf).

tsusanka, 2020-06-19:
Ah, right! Okay then. I have added a comment in 67b50f0fc to clarify and we can close.

closed 2020-06-19 by tsusanka