trezor/trezor-firmware #1073: Support Zcash Heartwood hardfork

tsusanka, 2020-06-18:
As confirmed via email, the consensus branch ID is the only change that will impact signing a transaction. We need to introduce a new `version` and set the consensus branch id to `0xF5B9230B`. Also I suppose the [fields](https://github.com/trezor/trezor-firmware/blob/e630279487dbdfefae14ceb962472531702cb026/core/src/apps/bitcoin/sign_tx/zcash.py#L56) we add for v4 should stay?

We should test this out on testnet where the hard fork allegedly already happened.

matejcik, 2020-06-18:
both `version` and `consensus_branch_id` are sendable by the user.
we might want to change `self.tx.version == 4` checks (in the core code) to `self.tx.version >= 4` to be future-proof against this kind of thing

tsusanka, 2020-06-18:
Ah, great!

> we might want to change self.tx.version == 4 checks (in the core code) to self.tx.version >= 4 to be future-proof against this kind of thing

Well what is the correct scenario to support Heartwood? To send `version = 4` and `branch_id = 0xF5B9230B` or support a new `version = 5`? The first option seems easier since it does not require any modifications in firmware?

matejcik, 2020-06-18:
We need to modify the firmware unfortunately, because higher versions are explicitly unsupported (wrt the "`version==4`" thing)

When that is done, the correct values to send are `version=5, branch_id=0xf5b9230b`
All further version bumps that do not touch the signature algorithm (ZIP243) will then work just by sending new correct values.

tsusanka, 2020-06-18:
Host needs to send `version = 4` and `branch_id = 0xF5B9230B` and that's it. No action in firmware is needed we need to do this in Suite/Wallet though.

tsusanka, 2020-06-19:
Confirmed via email that the version remains `4`. Therefore sending a new branch_id should be all that is needed. I am closing here as I have opened an issue in Suite https://github.com/trezor/trezor-suite/issues/2045.

closed 2020-06-19 by tsusanka