trezor/trezor-firmware #1078: Recovery of Monero wallet without Trezor device?

jonathancross, 2020-06-19:
**Is your feature request related to a problem? Please describe.**

If my Trezor is lost, damaged or fails, I would like an alternative way to recover my Monero wallet using the BIP39 seed phrase without a Trezor. Ledger provides [a python tool](https://github.com/LedgerHQ/app-monero/blob/master/tools/python/README.md) to convert their seed phrase backup into the 25 word seed phrase expected by the native Monero wallet software.

The assumption is that the user will have access to the identical wallet after importing the monero-specific 25 word seed into the GUI or commandline wallet.

**Describe the solution you'd like**

I would like an equivalent tool for the Trezor.

**Describe alternatives you've considered**

* Apparently @NoodleDoodleNoodleDoodleNoodleDoodleNoo built a [working POC](https://monero.stackexchange.com/a/1658/1023) based on `trezorctl` in 2016, but has since deleted the code.
* Ledger [seems to use](https://github.com/LedgerHQ/app-monero/blob/master/src/monero_key.c#L331) the same SLIP44 Cointype (`128`) at least, so their "Offline" conversion process might also work for a Trezor wallet as long as the [rest of the derivation path](https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/cli/monero.py#L22) is the same.

It would be nice to have both on-device and commandline options (like Ledger's tool), but the commandline option would probably be first priority and easier.

matejcik, 2020-06-22:
pinging @ph4r05 for his Monero knowledge.
however I don't see why the Ledger tool wouldn't work too.

ph4r05, 2020-06-22:
@matejcik thanks for a ping! I have some tools like this mentioned. I will check them and get back to you.

ph4r05, 2020-06-22:
@jonathancross @matejcik I've added some info on the seed recovery to the https://github.com/ph4r05/monero-agent/blob/master/PoC.md#trezor-seeds

I've authored a utility that can perform various seed conversions, also one mentioned in this issue: Trezor seed -> Monero seed. The utility: https://github.com/ph4r05/monero-agent/blob/master/monero_poc/seed.py

The scheme is quite simple:

```python
path = "m/44'/128'/0'"
master_seed = PBKDF2(password=MNEMONIC, salt="mnemonic"+PASSPHRASE, dkLen=64, prf=SHA512, iterations=2048)
monero_seed = SLIP0010("ed25519_seed", curve="ed25519", seed=master_seed, path=path)
monero_master = encodeint(decodeint(monero_seed)) # modular reduction by the curve order of the Ed25519
monero_spend_key = monero_master
monero_view_key = encodeint(decodeint(Keccak3_256(monero_master)))
```

The question is whether it is desirable to have this derivation extracted to some trezor python utility.

SLIP-0010: https://github.com/satoshilabs/slips/blob/master/slip-0010.md

tsusanka, 2020-07-02:
As already answered there are two ways you can achieve this:
- with the tools provided by @ph4r05
- build emulator (see our [docs](https://docs.trezor.io/trezor-firmware/core/build/index.html)) in case this is a real emergency

We do not want to include this into trezorctl not to increase its complexity. Thanks for the idea!

closed 2020-07-02 by tsusanka

jonathancross, 2020-07-08:
Thanks.
Is the Monero-specific recovery process documented for end-users someplace (other than this issue)?

tsusanka, 2020-07-08:
@jonathancross thanks I will forward it to our wiki team.

Dyrimon, 2022-02-03:
> @jonathancross it is not. Thanks, I will forward it to our wiki team.

Hi, I didn't find the instructions to build the emulator in the monero wiki. I had to dig it from the trezor forum
https://docs.trezor.io/trezor-firmware/core/build/emulator.html
https://docs.trezor.io/trezor-firmware/core/build/index.html

Please add the instructions in the XMR trezor wiki.

hynek-jina, 2022-02-03:
@Dyrimon Thank you,
We have noted it for the wiki improvements task list.

jonathancross, 2022-02-06:
The [instructions above](https://github.com/trezor/trezor-firmware/issues/1078#issuecomment-647647874) from @ph4r05  on how to convert a _24 word BIP39 seed phrase_ to a *_25 word Monero seed phrase_* are being added to the wiki?
https://wiki.trezor.io/Monero_(XMR)