trezor/trezor-firmware #1087: Introduce hard limit on transaction fee

tsusanka, 2020-06-30:
We [currently have](https://github.com/satoshilabs/trezor-firmware/blob/42eddf8e04c059832e0205925fd60f0943f3884b/core/src/apps/bitcoin/sign_tx/bitcoin.py#L134) soft warning when transaction fee seems to high. For a usual small segwit transaction this is  around 0.003 BTC which is ~30 USD at the current price (~9000 USD).

When there is a really big fee (by accident or malware) we are just a click away from spending a huge fee. As requested by Product, let's introduce a hard limit which is 10x more than the soft limit and may be turned of using advanced setting introduced in #1064. By hard limit I mean we refuse to sign and raise an error.

destrys, 2020-06-30:
I think this is a bad idea. 
This is another example on non-consensus opinions being baked into a hardware wallet.
How do you know that the fee market won't explode in the future, and someone who needs a next-block confirmation isn't willing to spend 0.03 BTC for that?

Instead of a built-in hard limit, I suggest having several warnings. So instead of a single click, it takes 3 or 4 clicks. 

Trezorctl is not a valid mode for turning off features, the majority of users are not comfortable using that route.

tsusanka, 2020-07-16:
> This is another example on non-consensus opinions being baked into a hardware wallet.

There are many things hardware wallets do that are non-consensus. BIP-39 is one of them (it is still "Proposed"). It is not the role of Bitcoin to decide these for us. We want to have as secure product as possible whilst allowing advanced users to turn off some checks. Feel free to contact me at tomas.susanka@satoshilabs.com if you wish to discuss this further, I do not want to make this issue a mess. Thanks!

tsusanka, 2020-08-03:
@mmilata would you be interested in a small research? Let's have a look on the current fee limits. That means create a table for top10 coins that we support (from coinmarketcap.com) and list for each one of them:
-

tsusanka, 2020-08-03:
@mmilata would you be interested in a small research? There are two things we would like to look at.

---

First let's have a look on the current fee limits. That means create a table for top10 Bitcoin-like coins that we support (or simpler way: use the ones that we currently support in wallet.trezor.io) and list for each one of them:
1. The current maxfee_kb.
2. The current fee_threshold for a "traditional" transaction (let's say something the user commonly deals with, doesn't have to be exact).
3. The current fee_threshold in USD. Basically (2) but in USD.
4. The hard limit we suggest (that is 10x (3) but it would be nice to see it in one table).

---

And second, let's investigate how we deal with fee thresholds in the other non-Bitcoin-like coins because I am not sure we do.

Do you think you could look at it at some point in future? The _Port_ has definitely a priority.

mmilata, 2020-08-03:
@tsusanka sure, can do that

mmilata, 2020-08-11:
@tsusanka For some of the bitcoin-based coins, the threshold for 250B transaction is following:

| Coin | `maxfee_kb` | `fee_threshold`        | USD rate | `fee_threshold` (USD) | hard limit (USD)         |
| :--- | ----------: | ---------------------: | -------: | --------------------: | -----------------------: |
| BTC  | 2000000     | 500000 sats = 0.005    | 11500.0    | 57.5                  | 575                      |
| BCH  | 500000      | 125000 sats = 0.00125  | 280.0      | 0.4                  | 4.0                      |
| BTG  | 500000      | 125000 sats = 0.00125  | 10.0       | 0.01                | 0.1                    |
| DASH | 100000      | 25000 sats = 0.00025   | 90.0       | 0.02                | 0.2                    |
| DGB  | 500000      | 125000 sats = 0.00125  | 0.03     | 0.00004             | 0.0004                 |
| DOGE | 1000000000  | 250000000 sats = 2.5   | 0.003    | 0.008                | 0.08                    |
| LTC  | 40000000    | 10000000 sats = 0.1    | 55.0       | 5.5                   | 55.0                       |
| NMC  | 10000000    | 2500000 sats = 0.025   | 0.4      | 0.01                  | 0.1                      |
| VTC  | 40000000    | 10000000 sats = 0.1    | 0.3      | 0.03                  | 0.3                      |
| ZEC  | 1000000     | 250000 sats = 0.0025   | 85.0       | 0.2                | 2.0                    |

Other coins such as ETH, ETC, NEM, XLM, ADA, XTZ don't have any warning or hard limit implemented.

tsusanka, 2020-08-17:
Closed by #1155 with two follow-up issues #1191 and #1192.

tsusanka, 2020-08-17:
### QA

Please test BTC transaction signing, it should not be affected.

bosomt, 2020-08-25:
maximum fee in suite is 2000 sat/b
i will check for another ways to test hard limit

bosomt, 2020-08-28:
**QA OK**

tested 2.3.3, no warning in older FW
@tsusanka is this feature core specific or is this legacy T1 included ?

FW version 2.3.3 df5421e
Wallet Git Commit: ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0

wallet log : 

["log","Fri, 28 Aug 2020 07:40:33 GMT","[trezor.js] [call] Sending","ButtonAck",{}]
["log","Fri, 28 Aug 2020 07:40:37 GMT","[trezor.js] [call] Received","Failure",{"code":"Failure_DataError","message":"The fee is unexpectedly large"}]
["log","Fri, 28 Aug 2020 07:40:37 GMT","[trezor.js] [session] releasing"]
["error","Fri, 28 Aug 2020 07:40:37 GMT","[AccountSendCtrl] Failed to send transaction","Error: The fee is unexpectedly large"]

[image]

tsusanka, 2020-08-28:
TT only for now. We will introduce it to T1 via _Port_.

bosomt, 2020-08-28:
tried Fee: $211.06

[image]
[image]

bosomt, 2020-08-28:
test results for Electrum 4.0.2:
1) treshold: same results, warning on device and in Electrum
2) hard limit:

[image]

maybe we will need fix for Electrum to properly handle FW error returned by device when you hit hard limit ? @matejcik

tsusanka, 2020-08-28:
This is a huge edge case let's not worry about that. (Unless this happens for every wire.DataErorr then it might be worth the fix.)

prusnak, 2022-03-21:
Let's implement this for Legacy too.

closed 2022-03-28 by hynek-jina