trezor/trezor-firmware #1088: Provide an option to retrieve the autolock delay setting

tsusanka, 2020-06-30:
To provide a good experience in Suite for https://github.com/trezor/trezor-suite/issues/1946 we need to publish the autolock delay setting in Trezor. Both in T1 and TT.

We might add it to Features but introducing something like `GetSettings` sounds more reasonable not to bloat them.

tsusanka, 2020-07-02:
Let's ask Suite team, because we do not care that much.

mroz22, 2020-07-07:
I dont like GetSettings very much for this. All other "options" are already present in Features.
Or do you mean introducing something like BasicFeatures containing only neccessary fields and then something like ExtendedFeatures with everything accessible via separate call? That sounds reasonable.

tsusanka, 2020-09-30:
Let's add it to Features the usual way for now. As discussed some time ago we can also approach the bloating features by answering Success to Initialize but that would require a breaking change. So let's postpone for now.

tsusanka, 2020-09-30:
@mmilata I think this one is a low-hanging issue as well.

PR #1295 (2020-10-02): Add auto_lock_delay_ms and display_rotation to Features
Fixes #1088.
closed 2020-10-02 by tsusanka