trezor/trezor-firmware #1093: Limit autolock delay setting - raise error if out of range

tsusanka, 2020-07-03:
We currently limit the auto lock to delay `min = 10s, max = 6 days` on TT and `min = 10s, max = ~50 days` on T1. In TT we silently set the appropriate min/max if the value is out of range. Let's raise an error in that case not to confuse the user. Also let's do that for Legacy if it is not there yet.

matejcik, 2020-07-03:
For the record, if you try to set less than minimum on TT, display confirmation dialog will show the minimum 10 s. This does not happen on the max side.

Also, T1 does not show the autolock value on screen. We might want to change that while we're at this.

tsusanka, 2020-07-03:
> Also, T1 does not show the autolock value on screen. We might want to change that while we're at this.

closed 2020-07-22 by tsusanka

mmilata, 2020-07-23:
Please test that setting the auto-lock delay to 5s causes an error.

tsusanka, 2020-08-05:
@sorooris for formal reasons could test this with the latest 2.3.2 RC please?

sorooris, 2020-08-05:
**QA OK**
tested with 2.3.2 cea6341
following message was displayed when setting values less than 10
```
trezorctl set auto-lock-delay 5
Error: ProcessError: Auto-lock delay too short

```

PR #1119 (2020-07-22): Autolock delay setting: raise error when out of range
Fixes #1093. The confirmation screen was changed to show the value in more human-readable units.