trezor/trezor-firmware #1095: RSOD after first unlock of 2.3.2 FW via PIn code

bosomt, 2020-07-03:
**Describe the bug**
Device crashed after I reconnected it and tried to enter PIN code

**Firmware version and revision**
2.3.2 revsion 14a8da9

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: web wallet 57337e4959a7d7db476a7ab5089febccea6f2c22
 - OS:nixos
 - Browser Chromium 
 - Version 83.0.4103.106

**To Reproduce**
i'm not able to reproduce it,
but i will try 

**Expected behavior**
device doesn't crash

**Screenshots**

[image]

**additional information**

```

trezorctl get-features
Features (188 bytes) {
    backup_type: Bip39 (0),
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Binance (3),
        Cardano (4),
        Crypto (5),
        EOS (6),
        Ethereum (7),
        Lisk (8),
        Monero (9),
        NEM (10),
        Ripple (11),
        Stellar (12),
        Tezos (13),
        U2F (14),
        Shamir (15),
        ShamirGroups (16),
        PassphraseEntry (17),
    ],
    device_id: '907F13CA441570E9ABF88DF1',
    flags: 0,
    initialized: True,
    language: 'en-US',
    major_version: 2,
    minor_version: 3,
    model: 'T',
    needs_backup: False,
    no_backup: False,
    passphrase_always_on_device: False,
    passphrase_protection: True,
    patch_version: 2,
    pin_cached: True,
    pin_protection: True,
    recovery_mode: False,
    revision: 7 bytes b'14a8da9',
    sd_card_present: False,
    sd_protection: False,
    session_id: 32 bytes 0x9c56b28d60cb856d48c4a6eca21ca5f80b2db6fe688d8012d7daf30eed437e4e,
    unfinished_backup: False,
    vendor: 'trezor.io',
    wipe_code_protection: False,
```

prusnak, 2020-07-03:
IMO there is only one place where this could've happened:

https://github.com/trezor/trezor-firmware/blob/25a2e690fee0e79e581863a946b030c77e9717d2/core/src/boot.py#L30

meaning that the exception was thrown somewhere from here:

https://github.com/trezor/trezor-firmware/blob/25a2e690fee0e79e581863a946b030c77e9717d2/core/src/boot.py#L16-L19

Assigning @matejcik and @andrewkozlik because they were the last to modify the suspected lines.

tsusanka, 2020-07-08:
@bosomt did you succeed to reproduce this?

bosomt, 2020-07-08:
@tsusanka nope ;(

tsusanka, 2020-07-16:
@bosomt still no luck? Did not happen for @wendys-cats or @sorooris ?

wendys-cats, 2020-07-16:
Confirming I have not been able to reproduce RSOD after reconnect with PIN enabled.

andrewkozlik, 2020-07-16:
I speculated that it might be a bug in the firmware update mechanism but I was unable to reproduce it or find anything in the code. Improved Mypy coverage might reveal what it is.

sorooris, 2020-07-17:
Confirming i wasn't able to reproduce this as well.

tsusanka, 2020-07-20:
@bosomt do you remember whether this has happened before, after, or during PIN entry?

---

Also @bosomt @sorooris could please try to retest this by:

1. Installing older fw (2.3.0 for example).
2. Setting up PIN.
3. Upgrade the device to current master (bcd044a).
4. Try to unlock the device and get address.

matejcik, 2020-07-20:
from slack:

￼> matejcik￼  13 days ago
> @bosoMT more details pls. to bylo zařízení který už bylo nainicializovaný? a tohle se stalo po reconnectu na screenu "not connected" po zadání pinu?
> a napodruhé to samý fungovalo? (edited) 
> ￼￼
> bosoMT￼  13 days ago
> ano bolo inicializovane
> bolo to po reconnecte,
> potom sa to uz neopakovalo ￼

bosomt, 2020-07-20:
@tsusanka 
- I've never seen this error before or after this one accident
- 2.3.0 unsigned to current master so device is not wiped ?

bosomt, 2020-07-20:
tested 

trezor-fw-regular-2.3.0-0b7a8449.bin 
>
trezor-fw-regular-2.3.2-bcd044a4.bin

works OK
i will try some other operations

tsusanka, 2020-07-20:
> I've never seen this error before or after this one accident

I meant whether this has happened to you before you have tried to enter PIN or **during** or **after** you have entered PIN. In another words when exactly did it happen. If you remember of course :).

>  2.3.0 unsigned to current master so device is not wiped ?

Yes please.

bosomt, 2020-07-20:
@tsusanka i think it was hen i entered and confirmed PIN when I reconnected device, if I remember it correctly :) it was 9PM :)

tsusanka, 2020-07-24:
Please reopen in case you (or anyone) reproduce this.

sorooris, 2020-08-05:
Reproduced successfully by accident with 2.3.2 rev cde8f31e
[image]
Wallet: develop 80096add47b6b06e726f1650970710d4fa2ccb3e
Will try to reproduce again.

tsusanka, 2020-08-05:
@sorooris did you have a clean device (no firmware) or did you have something installed previously?

sorooris, 2020-08-05:
Device had previously the 2.3.2 revision 731e9b7c installed.

sorooris, 2020-08-05:
I was able to reproduce the RSOD again.
I still need to verify if it was just luck or pure magician skills. I'll keep you updated with steps in a minute.
[image]

sorooris, 2020-08-05:
I was able to identify how to reproduce this RSOD.

Steps to reproduce:
1. connect a previously disconnected device to computer
2. wait the time until the red warning screen is displayed
3. start tapping the device screen frantically and as fast as possible

Bug happens right after the last step.

matejcik, 2020-08-05:
good job triggering this! a fix is coming shortly

The problem is that if a `TOUCH` event arrives before a `REPAINT` event (which is under normal circumstances the first thing that happens), the homescreen class is not fully initialized yet.
It probably wouldn't happen in a production FW due to the introductory red screen missing, but still, good to squash this one.

matejcik, 2020-08-05:
could you retest with firmware from [here](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/174555160)

closed 2020-08-05 by matejcik

sorooris, 2020-08-05:
**QA OK**
Couldn't replicate the bug however frantically I tried to tap the screen :smile: 
Tested with the 2.3.3 build revision cea63415.

commit cea63415 (2020-08-05): core: make sure Homescreen is properly initialized (fixes #1095)