trezor/trezor-firmware #1099: Soft auto-lock feature interrupts dry run recovery

wendys-cats, 2020-07-07:
**Info:**
 - **OS:** nixOS
 - **Browser:** Chromium Version 83.0.4103.61 (Developer Build) (64-bit)
 - **Wallet Version:** **suite** suite-web/feat/232-pinentry 06d90d21312e090fb2e1b14f83b99be48bd13fb2 AND  **wallet** autolock/development 
 - **Firmware Version:** 2.3.2 b14a8da9
 
**Describe the bug**
New auto-lock-delay feature does not recognise dry run recovery mode and will soft-lock the device duing it, canceling the action. Happens both in suite (pinentry branch) AND wallet (autolock/development branch)

**Steps to reproduce:**
1. Connect device
2. Set PIN, set auto-lock-delay (to lower numbers, f.e. 10/15/20)
3. Start dry run recovery

**Expected result:**
New feature will not interrupt running seed check.

**Actual result:**
Device does lock, and cancels the seed check (even in the middle of typing a seed word)

tsusanka, 2020-07-07:
This is a firmware issue. We have discussed on today's standup we would like to fix this by locking only on homescreen. Let's discuss this tomorrow with @matejcik.

tsusanka, 2020-07-07:
This issue also applies for an actual recovery/backup (not a dry run).

tsusanka, 2020-07-07:
On a quick test this does not seem to affect tx signing.

matejcik, 2020-07-07:
> This is a firmware issue. We have discussed on today's standup we would like to fix this by locking only on homescreen. To be precise we will calculate the delay all the time but do the actual locking on homescreen (= after the workflow). Let's confirm this on tomorrows standup with @matejcik.

we have previously discussed that autolock should actually interrupt actions, iirc?
it makes sense: if you keep a signtx operation open for 10 minutes, it should interrupt (consider a user that has to run outside while waiting for tx streaming to finish)

matejcik, 2020-07-08:
> On a quick test this does not seem to affect tx signing.

In my testing, TX signing is cancelled and locked after a period of inactivity.

Note that USB communication does count as activity. Streaming previous txes for 30 seconds will not let the device lock, but waiting 10 seconds on a confirmation screen will.

tsusanka, 2020-07-08:
> Note that USB communication does count as activity.

Ah I haven't realized that.

closed 2020-07-10 by matejcik

wendys-cats, 2020-07-13:
`QA test successful `

**FW** 2.3.2 revision bb8127c5
**Wallet branch** /autolock/development

Seed check bug fixed in #1102.  Actual backup/recovery is NOT a part of #1102 - applies only for dry run.

Can confirm dry run is no longer interrupted mid-typing. Right now, seed check and backup work the same - after waiting for X amount of seconds, auto-lock will apply and interrupt the action.
Also, Error message added when trying to set auto-lock with disabled PIN.

sorooris, 2020-07-15:
I don't understand why was this issue moved to the Merged column of the board - should it be tested again? Please let me know @matejzak @tsusanka

matejzak, 2020-07-15:
@sorooris I was unable to move this out of the Suite July 8 project (probably since I didn't put it there in the first place). It this has been tested before, we are good, no need to test it again.

sorooris, 2020-07-15:
sweet, thanks!

tsusanka, 2020-07-15:
This was originally created in the Suite repo but it was a firmware bug so I have [moved](https://github.com/trezor/trezor-firmware/issues/1099#event-3520810846) it here. Maybe that's why it ended in the Suite project as well. Tl;dr: firmware bug -> firmware process -> QA test successful label present -> all good :).

commit daec5dde (2020-07-10): core: touch idle timer in keyboards (fixes #1099)
PR #1102 (2020-07-10): Ping idle timer while in recovery mode
fixes #1099 by making sure the device does not auto-lock while entering a word.

also adds a warning to trezorctl that you don't have PIN and you're trying to set up autolock. should help testers out

---

Also adds a little tool that can record and generate a click-test case from a trezorctl sequence.
It's a horrible thrown-together hack, and at this moment I'm not interested in improving the code of this tool even a little.

If you feel that a tool in this state should not be commited into our repository, I'll remove it from the PR and publish separatel [... cut]
commit 05176357 (2020-07-10): core: touch idle timer in keyboards (fixes #1099)