trezor/trezor-firmware #1103: non-mandatory-script-verify-flag (Signature must be zero for failed CHECK(MULTI)SIG operation)

KayBeSee, 2020-07-10:
I am building a multisignature wallet and am running into a weird issue when signing with Trezors.

I can sign the following transaction with Ledgers and Coldcards and they are able to broadcast them no problem but when I sign with a Trezor, I get the error below.

Transaction
```
020000000001012ab7b563a9ff24b0fd0e8b618aa1fe64d544be9bc80bed18c2d23e298a1426630100000000ffffffff02801a06000000000016001461617434ee35dc8b7d3492796a97a7f2a8642c172a460100000000002200207f43cfb165236ddae6bc7cfcebfd6ac51d6b39fa90ac6064ccc8da90f20214ef0400473044022041a374d73cc630832ce36e1a45fbadda07280db8fdf104c225c6118062bc7c9a02206a907aead8b964e396fe7d9299c8b820494844b200994cd5898f84c30c3ed38901483045022100d96a4ff0798ff1713515756c781d99aa90894e0be99fbd861f77d80dc56fd739022025e038ed376ef6d4e9777f95e4f994497240d41ab645e04187814a92b3ac7e3a016952210200fc2d0e055817fe1dbeb4c2f596c4688d03c181fc1179ae312ad24430ed30d021023d993df44e953860622be1667bf360e992236c1037e570da309cdb7e6a10afdb210374bcb3947992d08817edd387e66d0aad4b13f07bd29d9eda136bd36ea2036bd853ae00000000
```

Error
```
{
"code": -26,
"message": "non-mandatory-script-verify-flag (Signature must be zero for failed CHECK(MULTI)SIG operation)"
}
```

I understand that with the most recent firmware update, I need to attach the previous transaction to the nonWitnessUtxo field in the input, which I am doing. It looks like it is signing it no problem on the device, but when I attempt to broadcast it fails.

I looked online and it said it might be a precision error with giving the wrong values on the input but I don't think that is the case here since I can sign and broadcast with other hardware wallets.

Any ideas?

prusnak, 2020-07-13:
Please describe more in detail how are you sending TX to Trezor. Which libraries/programs do you use, show us the code, etc.

prusnak, 2020-07-13:
Might be relevant: https://github.com/bitcoin-core/HWI/pull/352/files

Which version of HWI and Bitcoind do you use?

matejcik, 2020-07-13:
@KayBeSee please provide the transaction you are signing, preferably in the `trezorctl` [json format](https://github.com/trezor/trezor-firmware/blob/master/python/docs/transaction-format.md)

KayBeSee, 2020-07-14:
I am using HWI 1.1.2 and broadcasting the transaction using Blockstream's API.

Here is the hex of a transaction I was trying to sign and broadcast earlier:

```
020000000001012ab7b563a9ff24b0fd0e8b618aa1fe64d544be9bc80bed18c2d23e298a1426630100000000ffffffff02801a06000000000016001461617434ee35dc8b7d3492796a97a7f2a8642c172a460100000000002200207f43cfb165236ddae6bc7cfcebfd6ac51d6b39fa90ac6064ccc8da90f20214ef0400473044022041a374d73cc630832ce36e1a45fbadda07280db8fdf104c225c6118062bc7c9a02206a907aead8b964e396fe7d9299c8b820494844b200994cd5898f84c30c3ed38901483045022100d96a4ff0798ff1713515756c781d99aa90894e0be99fbd861f77d80dc56fd739022025e038ed376ef6d4e9777f95e4f994497240d41ab645e04187814a92b3ac7e3a016952210200fc2d0e055817fe1dbeb4c2f596c4688d03c181fc1179ae312ad24430ed30d021023d993df44e953860622be1667bf360e992236c1037e570da309cdb7e6a10afdb210374bcb3947992d08817edd387e66d0aad4b13f07bd29d9eda136bd36ea2036bd853ae00000000
```

I am building the transaction using bitcoinjs-lib. I can share that code if it is useful.

tsusanka, 2020-07-20:
@KayBeSee could you please provide the json format for trezorctl as @matejcik mentioned? Or PSBT. That would help us a lot to debug this issue.

KayBeSee, 2020-07-21:
PSBT.toBase64 (this is what I am passing to HWI for signing):
`cHNidP8BAH0CAAAAARYK0qAl5LSN4zAlBJLV2hzQcMoNvVc6ep0E8Tq6FWObAQAAAAD/////AqCGAQAAAAAAFgAUDNMQS2XovLv37nK58I3ZyGjPB49REwQAAAAAACIAILtt9eLCTm9k8wZAd3HzistOcG55NWMISGmTBU7E49RaAAAAAAABAP18AQIAAAAAAQEqt7Vjqf8ksP0Oi2GKof5k1US+m8gL7RjC0j4pihQmYwEAAAAA/////wKghgEAAAAAABYAFGFhdDTuNdyLfTSSeWqXp/KoZCwXodsFAAAAAAAiACB/Q8+xZSNt2ua8fPzr/WrFHWs5+pCsYGTMyNqQ8gIU7wQARzBEAiAtbwkSqGkKTnodgOV996E2+9DmMkv8u1O2OBuoZcSjOgIgE68iuo+ilmvIRVrf4i51l8o9pugFj2iEMdnkAIfzIc0BSDBFAiEAyyp41SfNvXlkdU3oNBNQ/NTFHd+52FiFhlu2G1PkjnACIAW3YopAUZCkr6jtbElcV7faOhtZE+vbaXB6XpwTlKaNAWlSIQIA/C0OBVgX/h2+tML1lsRojQPBgfwRea4xKtJEMO0w0CECPZk99E6VOGBiK+Fme/Ng6ZIjbBA35XDaMJzbfmoQr9shA3S8s5R5ktCIF+3Th+ZtCq1LE/B70p2e2hNr026iA2vYU64AAAAAAQVpUiECeYydan/pZhgn2O1U4b7496G81PsppdzpcAcYWm4QgRYhAsZeeeZRkko/NNG79+CPe/h6Gax5Gcdij8CpZA5KGmLAIQOicGfp2ZBljVDMBLIQHfu59JeC4mp3ZOfmrWBxa2KRPFOuIgYCeYydan/pZhgn2O1U4b7496G81PsppdzpcAcYWm4QgRYctPOWgTAAAIAAAACAAAAAgAIAAIABAAAAAgAAACIGAsZeeeZRkko/NNG79+CPe/h6Gax5Gcdij8CpZA5KGmLAHMHL+U0wAACAAAAAgAAAAIACAACAAQAAAAIAAAAiBgOicGfp2ZBljVDMBLIQHfu59JeC4mp3ZOfmrWBxa2KRPByRMMPWMAAAgAAAAIAAAACAAgAAgAEAAAACAAAAAAAA`

JSON (I constructed this manually by just copying the values into the right format):
```
{
  "coin_name": "Bitcoin",
  inputs: [
    {
      address_n: m//48'/0'/0'/2'/1/2,
      prev_hash: 9b6315ba3af1049d7a3a57bd0dca70d01cdad592042530e38db4e425a0d20a16
      prev_index: 1,
      script_type: 3,
      sequence: 4294967295
    }
  ],
  outputs: [
    {
      address: bc1qpnf3qjm9az7thalww2ulprweep5v7pu0v9ml6q,
      amount: 100000,
      script_type: 4
    },
    {
      address: bc1qhdkltckzfehkfucxgpmhruu2ed88qmnex43ssjrfjvz5a38r63dqfu7ffu,
      address_n: m/48'/0'/0'/2'/1/3
      script_type: 4,
      amount: 267089
    }
  ],
  details: {
    version: 2,
    locktime: 0
  }
}
```

After reading through the documentation for trezor JSON format I noticed that I am not appending the amount, address_n, nor script_type to the inputs (they are not [supported types in Bitcoinjs-Lib](https://github.com/bitcoinjs/bip174/blob/master/src/lib/interfaces.d.ts#L26). Do you think that could be causing the issue?

prusnak, 2020-07-22:
> Do you think that could be causing the issue?

I would say so ...

closed 2020-07-22 by prusnak