trezor/trezor-firmware #1108: do not gc.collect so often in Bitcoin signing

matejcik, 2020-07-14:
related to #1038 

At this point it *seems* that we have quite enough free memory for Bitcoin signing that the normal GC collector can handle it fine by itself.
Calling `gc.collect` explicitly after every TxRequest/TxAck exchange is therefore wasteful and [noticeably](https://github.com/trezor/trezor-firmware/pull/1089#issuecomment-656637114) slows down the process.

Similarly, we probably don't need to wrap UI operations in unimports.

tsusanka, 2020-07-22:
@mmilata I think this can be your next issue when the time comes :). @matejcik am I correct that we want to remove **all** gc.collect calls in `apps/bitcoin/`? Or maybe only the ones in `helpers.py`?

matejcik, 2020-07-22:
definitely the ones in `helpers.py` that happen after every message exchange.

not sure about others, some review will be necessary

tsusanka, 2020-07-24:
Closed by #1128, unimports moved to #1132.

closed 2020-07-24 by tsusanka

tsusanka, 2020-08-05:
QA has tested this as part of the freeze testing.