trezor/trezor-firmware #1109: Stream prev_tx after confirmation

andrewkozlik, 2020-07-14:
The streaming of previous transactions can get quite lengthy when there is a large number of transaction inputs, see https://github.com/trezor/trezor-firmware/issues/1038. Currently we stream the previous transactions before the user confirms anything, which means that when the user wants to sign a large transaction, they have to:

1. Wait for the previous transactions to get streamed.
2. Confirm the transaction outputs and the total amount.
3. Wait for the signing to take place.

We want to switch phases 1 and 2 to improve the UX. Possible disadvantages:
- There will be one additional `TxRequest` per input, which will make the signing take even longer overall, but the difference is probably going to be negligible. 
- It will require the amount field to be supplied always for every input, which might break some 3rd parties.

My plan is as follows:

- [ ] Implement this in TT.
- [ ] Hand the firmware over to QA to determine whether any 3rd parties will be broken due to the amount field requirement.
- [ ] Evaluate whether/when we want to release this change and inform 3rd parties.
- [ ] Implement in T1.
- [ ] Fix device tests.
- [ ] Code review.

andrewkozlik, 2020-07-15:
I implemented this in the TT firmware and fixed the device tests already but haven't tested it with any wallet.
Branch: https://github.com/trezor/trezor-firmware/compare/andrewkozlik/prevtx-after-confirm.
Firmware: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/639680430/artifacts/file/trezor-fw-regular-2.3.2-d065c5df.bin

@tsusanka please hand this over to QA and give them a list of wallets/applications we want to test against. After testing bitcoin, it would be good to also test zcash, decred, bcash, bgold with at least one wallet.

tsusanka, 2020-08-06:
I have run a quick test of this and it actually fails on Wallet as well. On the "Missing amount field" error. I let QA test the 3rd parties (Electrum, MyCelium and Exodus) but I guess it will break them as well.

bosomt, 2020-08-15:
- wallet **OK**
- suite **OK**
- Electrum 4.0.2 **OK**
- Exodus **not tested yet** , they don't have custom fees so I need to sync with rest our team so we wont spend 20$ to test one opp ;)

more details on ontestpad

andrewkozlik, 2020-08-15:
Strange, why are you @bosomt and @tsusanka getting different behavior for Wallet? Best to double-check you have the right firmware, it should be d065c5d. There is also a chance that wallets might behave differently for segwit accounts and for legacy accounts.

bosomt, 2020-08-15:
get features says:     revision: 7 bytes b'd065c5d',

BUT

first i was not able to load wallet at all
crashed during loading with this error:

[image]

tsusanka, 2020-08-16:
> There is also a chance that wallets might behave differently for segwit accounts and for legacy accounts.

@bosomt I have indeed tested a non-segwit inputs. Please test against Wallet again once with Segwit input and once with an old non-Segwit.

bosomt, 2020-08-16:
@tsusanka 

I'm sorry but all out legacy accounts are empty :(
can you we test it via testnet ? or can you arrange some inputs for us  ? :)

andrewkozlik, 2020-08-16:
Testnet should be fine.

bosomt, 2020-08-17:
yes it fails with legacy coins

wallet: 
**Failed to send transaction
Error details: Missing amount field.**

i will test rest of wallets

bosomt, 2020-08-17:
legacy:

Wallet: **NO OK**
Suite **OK** commit https://github.com/trezor/trezor-suite/commit/5e63937356f4dc5c806560e74f6c926ec156f840 https://tbtc1.trezor.io/tx/0b3ab259dbbc69a75a3a20269deb1778c1ab63f427928c4624fadb04c244eac4 
Electrum **in progress**
Exodus: **NA** no testnet or legacy inputs
Mycellium **NA** no testnet support or legacy inputs

sorooris, 2020-08-18:
Results:
**2.3.2** revision d065c5df

- Wallet (MyTrezor, public): **OK**
- Suite (beta): **OK**
- Electrum (4.0.2): **OK**
- Exodus (20.1.30): blocked by tx fee and most of the funds on the second account which the app does not read.
- Mycelium: had no Android device available.

Detailed test report available [here](https://satoshilabs.ontestpad.com/script/178/report/?auth=98683e5dbea2c3df7808d2a27a522374).

tsusanka, 2020-08-18:
@sorooris please investigate why you and @bosomt have different results for "send legacy BTC tx".

andrewkozlik, 2020-08-18:
There is probably a mixup in what you guys are testing. When I mentioned segwit accounts and legacy accounts, I meant these to be the source accounts, from which you are sending the transaction. The destination account type shouldn't influence the behavior.

@sorooris, @tsusanka noticed that the transaction for Wallet send legacy BTC tx and Suite send legacy BTC tx the transaction is the same (https://btc1.trezor.io/tx/89382e464915464b73c6c8575e699f8d616d89566c2de828e37249617c3ae5ae).

andrewkozlik, 2020-08-18:
So I looked at your transations:

@sorooris 
1. Wallet: both transactions look like they are from a segwit account. However, see mixup mentioned above.
2. Suite: again both look like they are from a segwit account (one native, the other P2WPKH in P2SH). However, see mixup mentioned above.
3. Electrum: looks good.

@bosomt
1. Wallet segwit tx: looks good.
2. Suite: both transactions look like they are from a legacy account.
3. Electrum looks good.
4. Mycelium looks good.

It turns out you managed to overlap each others mistakes so nicely that when I put all the results together, it looks like the only thing that nobody has been able to do so far is a transaction from a legacy account in Wallet.

andrewkozlik, 2020-08-19:
So I did a little digging in the wallet code and it looks to me like a one-liner will get this working, but I need someone to compile and test it: I found that `buildTx()` in `hd-wallet` allows you to build a transaction without input amounts by specifying `inputAmounts=False`. I don't understand why anyone would want to do that, but I guess we can keep that code. The place to fix it IMHO is to set `const inputAmounts = True;` in `doBuildTx()` here: https://github.com/satoshilabs/mytrezor/blob/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0/app/scripts/trezor/TrezorAccount.js#L884

While looking at this I noticed a recent change in mytrezor/app/scripts/trezor/TrezorAccount.js that I am not entirely happy about, https://github.com/satoshilabs/mytrezor/commit/2cbbb5813d108f29b307d7b25c97a2a40cb9ae30. I think the correct fix at the moment would be for `prevTxRequired()` to return True always, because eventually we will want this behavior for taproot. But maybe this code will be deprecated by then? BTW, a code comment about segwit was left in TrezorAccount.js, which is no longer true.

andrewkozlik, 2020-08-19:
I created a pull request https://github.com/satoshilabs/mytrezor/pull/949.

sorooris, 2020-08-20:
@andrewkozlik Thank you so much for looking into the details; I re-ran the test for the legacy to legacy transaction in production Wallet and the issue is indeed present but just in case I'll re-run the tests again for both native and p2wpk-in-p2sh in other apps as per the Slack thread.

**Test NOK**
Wallet: [production](https://github.com/satoshilabs/mytrezor/commits/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0)
Device: model T 2.3.2 d065c5d
Transaction details:
> Final fee: 2 sat/B 
> Size: 0.195 kB 
> Fee: 0.0000039 BTC
> Inputs:
> [c62262346d6e9856267b8740e0c3cadbd1a16a7da4944f4ba95927f8f2f95668:0](https://btc3.trezor.io/tx/c62262346d6e9856267b8740e0c3cadbd1a16a7da4944f4ba95927f8f2f95668)
> Outputs:
> 0.00008118 BTC 
> 1773M3sfbQ2Y89XTjhVVHByXc6WetjLrgN

[image]

tsusanka, 2020-08-31:
**Product stance:** Just for the record the Product agrees with it. Slush even mentioned this is how it used to be in the past.

andrewkozlik, 2020-09-07:
Merged into master.

@qa Please wait for the fix satoshilabs/mytrezor#949 before testing with the Trezor web wallet. Test with T1 and TT. In addition to bitcoin, it would be good to also test zcash, decred, bcash, bgold with at least one wallet.

closed 2020-09-07 by andrewkozlik

tsusanka, 2020-09-24:
Tests from @bosomt:

OK  T1 1.9.4
mytrezor OK https://btc3.trezor.io/tx/55a400ba076a3bf8b21dd14bbbcc5d7331ae56a2295a1fa2f21233dd91f44631
suite 20.9.1 OK
https://btc1.trezor.io/tx/725d80a009344cffdcff2ef27385983ddcb567c2f443cbcd8a6a9802195af32a
electrum 4.0.2
https://btc1.trezor.io/tx/0f9a30ab2d2a6365f530b5184e80a99a7d77d95200b52f4dc4df07da0dc1860a

OK TT 2.3.4
mytrezor
https://btc3.trezor.io/tx/b68dd25cc26c61d8cd705b4706aa140cc7a8228ad46f55d9bb3f6a96d8ac78e1
suite 20.9.1
https://btc1.trezor.io/tx/febacfb4fc16a4179a0d923a4382c853d3f28840aed3d30aea3b7a748da03d95
electrum 4.0.2
https://btc1.trezor.io/tx/240134151add34ad0d54b7fb4c257de3c9b4b0117d930d3805a6c71467dc5243

Legacy:
1.9.4 Legacy mytrezor
https://btc3.trezor.io/tx/8ceb5de5143c97bc588b087c1e338ab5328386193a7f4594a2bb278b6e82d560
2.3.4 Legacy mytrezor
https://btc3.trezor.io/tx/9dffe36164b95b407b2975c3d87e919ce0546f4f348552bf6ba76115c2474bcb

bosomt, 2020-09-24:
BCH mytrezor 

1.9.4 https://bch3.trezor.io/tx/7f92d66827d40c3067db8b53df0e7d09e921cb5cc45ae3862e22f363c98e5727
2.3.4 https://bch3.trezor.io/tx/1289c143bf4d158ad7f6165f0e9be9275810fa57081cf76c682c5f644a0a1931

ZEC mytrezor

1.9.4 https://zec3.trezor.io/tx/1912f3aa1c8c9008aafe3483ace848a70d800a0f41c80f15a9aca0ba8aa60ee6
2.3.4 https://zec3.trezor.io/tx/a0c8fb11a07ccabb27dd8376d665cf6dd3cd205def02f39ad7e02b9c3b424528

as soon we receive decred from exchange we will test it via exodus

tsusanka, 2020-09-25:
From @wendys-cats:

2.3.4 legacy mytrezor BTC
https://btc3.trezor.io/tx/62cde1bd82774d663e4d4ae0c1071f00b8f864b68da985042296634a043c4dad
2.3.4 legacy mytrezor BCH
https://bch3.trezor.io/tx/50eef1175c3459d9a7beedccc09fd7ac64cfa6674dfcf25166cb401385ba8372
2.3.4 legacy mytrezor ZEC
https://zec3.trezor.io/tx/6217c2c1466367462ce8d095358073a038f4d160fb522f2924e3420360c426d9

1.9.4 legacy mytrezor BTC
https://btc3.trezor.io/tx/a2d0334b72d9e0eb5fe699c2f7b2d20a3d1a6d768e92c966c1811db59e3cf417
1.9.4 legacy mytrezor BCH
https://bch3.trezor.io/tx/5ca6470df728cba87302c08317169d9485a97c56243e56ccb7570b545e339d0d
1.9.4 legacy mytrezor ZEC
https://zec3.trezor.io/tx/0d019a607a9376e8730e07899ccdd0dfd185f8e0b904a97bcb2ebe7532ff47e2

Looks like it is working!

andrewkozlik, 2020-09-25:
Tests look good to me.

For the record, this one was actually sent from a segwit account: 
> 1.9.4 legacy mytrezor BTC
> https://btc3.trezor.io/tx/a2d0334b72d9e0eb5fe699c2f7b2d20a3d1a6d768e92c966c1811db59e3cf417

but there is no need to retest, since this one already covers it:
> Legacy:
> 1.9.4 Legacy mytrezor
> https://btc3.trezor.io/tx/8ceb5de5143c97bc588b087c1e338ab5328386193a7f4594a2bb278b6e82d560

tsusanka, 2020-09-25:
The revision tested was f8489f16.

bosomt, 2020-09-27:
no ide if this is problem with FW or Exodus but got this error when i tried to send Decred 
fw 2.3.4 commit feb24e4

{"name":"RangeError","message":"The value of \"offset\" is out of range. It must be >= 0 and <= 245. Received 246","stack":"RangeError [ERR_OUT_OF_RANGE]: The value of \"offset\" is out of range. It must be >= 0 and <= 245. Received 246\n at boundsError (internal/buffer.js:73:9)\n at Uint8Array.readUInt8 (internal/buffer.js:239:5)\n at e.i.readUInt8 (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:7:367494)\n at Function.l.fromBufferReader (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:13:434409)\n at E.fromBufferReader (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:79:48743)\n at E.fromBuffer (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:79:48532)\n at new E (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:79:43280)\n at Object.fromBuffer (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:206830)\n at Function.fromBuffer (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:86:559921)\n at Object.e.exports [as signBitcoinTransaction] (/nix/store/40x4cr8skdhrjc2h8xwq0yycm68qzbfh-exodus-20.1.30/resources/app.asar/src/app/vendors.js:180:205866)"}

andrewkozlik, 2020-09-28:
@bosomt, I can't tell what the problem with Exodus + Decred is. Please try the same with the last stable FW release 2.3.3.

bosomt, 2020-09-29:
@andrewkozlik @tsusanka **works OK** 

DCR is working fine via latest Exodus 20.9.28 Windows10 and 2.3.4 55baf3ac, i was able to send and receive transaction

problem was that Nixos uses Exodus that's more than 6 months old