trezor/trezor-firmware #1114: model T v2.3.2 responds with firmware error when loading white noise homescreen

bosomt, 2020-07-20:
**Describe the bug**
model T with 2.3.2 responds with firmware error when loading white noise homescreen
`Homescreen change failed.
Error details: Firmware error`
Default homescreens: `Success
Homescreen has been changed.`

with production FW 2.3.1 works fine

**Firmware version and revision**
FW 2.3.2 Revision e70811c

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: web wallet b3f18eb73765f92dd35eff2c562ef1edb5dd9fa3 
 - OS: Nixos
 - Browser CHromium
 - Version 83.0.4103.116

**To Reproduce**
Steps to reproduce the behavior:
1. wallet/suite/trerzorctl 
2. connect device with 2.3.2 revision e70811c
3.try to load custom background
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

wallet log: 
["log","Mon, 20 Jul 2020 12:17:16 GMT","[trezor.js] [call] Received","Failure",{"code":"Failure_FirmwareError","message":"Firmware error"}]
["log","Mon, 20 Jul 2020 12:17:16 GMT","[trezor.js] [session] releasing"]

trezorctl error:

ash-4.4$ trezorctl set homescreen -f ./Downloads/white-noise.toif 
Error: FirmwareError: Firmware error
bash-4.4$ trezorctl set homescreen -f ./Downloads/white-noise.toif 
Error: FirmwareError: Firmware error

bosomt, 2020-07-20:
works ok with current master
trezor-fw-regular-2.3.2-bcd044a4.bin

matejcik, 2020-07-21:
it seems that we're (still) running out of memory.
@bosomt please check if the issue persists if you do this on a freshly plugged-in Trezor

matejcik, 2020-07-21:
if not, please continue testing with the whitenoise background set

bosomt, 2020-07-21:
**QA OK**

@matejcik 
works fine now with same commit e70811c
i wiped device to bootloader, loaded FW and created new seed
do you want to close this issue ?

trezorctl set homescreen -f ./Downloads/white-noise.toif
Please confirm action on your Trezor device.
Settings applied

suite: Settings changed successfully

matejcik, 2020-07-21:
i'll lower severities for now and see what we'll do about this

tsusanka, 2020-07-24:
Related to #590.

bosomt, 2021-02-01:
@tsusanka @matejcik 

we are running out of memory again, when you set white noise Homescreen device will freeze after load.
Device will freeze when you try to sign BTC tx later or when you try to load white noise homescreen again.
Homescreen size: 63400 B

FW 2.3.5 commit ffa96205f

**screenshots**

$ trezorctl set homescreen white-noise.png
Error: DataError: Failed to decode message: memory allocation failed, allocating 16849 bytes

[image]

[image]

guiambros, 2021-02-08:
Same here, with a fresh new Trezor T. It fails when trying to upload a custom homescreen, both jpg as well as png 144x144. It returns "Firmware error", both through Trezor Suite and wallet.trezor.io. I'm using stock firmware 2.3.4.

[image]

PR #1565 (2021-05-06): App separation and workflow restarts
should fix #1114

also fixes #946 

I recommend you review this not as a whole but going by individual commits. That way it's much more readable what is going on:

5fa959a27e38680a31fd556804b82f9e6b7a62f1 ensures that buffers in `trezor.wire` are allocated at import time, i.e., in a more predictable place

aa2504dba5ed27da9815a3bacd12b8e3e3e33c3f just moves `empty_bytearray` to `trezor.utils` where it actually fits much better

d6d4c0d9af39815f33f483645e2e0d5ca32e03f9 refactors `storage.cache` to preallocate all the space it needs at import time. It also mo [... cut]
closed 2021-05-06 by matejcik