trezor/trezor-firmware #1117: Hedera (Hbar) Integration

cacampbell, 2020-07-20:
- What is the approximate scale? Have a look at the other apps and try to compare your new app with one of those present in the terms of scale.

> Around the size of the ripple app 

- Do we need any new cryptographic primitives to support this coin?

> no

-Is there (or will you develop) a frontend wallet that will support Trezor with this coin?

> yes, https://myhbarwallet.com

Provide links to relevant developer documentation, esp. about address format and transaction signing

> https://docs.hedera.com/guides/

Provide a link to a transaction explorer

> https://explorer.kabuto.sh/

___

Previously, we tried adding support here:

- https://github.com/trezor/trezor-firmware/pull/502
- https://github.com/trezor/connect/pull/449/files

Now that these pull requests are out of date, and hedera has made it to #29 on coinmarketcap, we would like to try again. 

Thanks.

prusnak, 2020-07-21:
Hedera is # 51 on CMC

closed 2020-07-21 by prusnak

cacampbell, 2020-07-21:
[image]

cacampbell, 2020-07-21:
Ah, the homepage applies the coins only filter ¯\_(ツ)_/¯

woops well we'll wait until hbar is worth 5 cents then I guess.

scarycz, 2021-01-29:
Hello @prusnak , can you please clarify? I'm waiting for Trezor to support Hedera for quite a while and second year I'm seeing the same approach taken (#502). Either suddenly raising the onboarding bar or completely ignoring the development the project has taken since. It makes me curious how were all other claimed 1617 coins & tokens selected? All of them made it to top 30 or I just selected wrong wallet?

prusnak, 2021-01-29:
> It makes me curious how were all other claimed 1617 coins & tokens selected?

Most of these are tokens and they are trivial to be added. If hedera was a token it would be added easily too. Since it is not and requires a lot of work to be integrated, this work needs to be justified by being in top30.

scarycz, 2021-02-16:
In this particular case, some foundation has been already laid out in the PR referred..
Based on current and previous reactions seen, I won't waste too much energy on getting deeper. From very shallow consumer point, I could list coins which are not included from top 30, I could also list coins supported and being far below the minimum bar. Would it matter? Probably not.
What makes me curious whether this single and pretty tight criteria doesn't seem obsolete considering the dynamic market it's supposed to facilitate? Aren't there also other performance indicators? Seeing above example, you are potentially leaving share of projects who are willing to invest development effort left off without proper evaluation