trezor/trezor-firmware #1118: core: invalid homescreen leads to bootloop

matejcik, 2020-07-21:
to reproduce:

use the following file
[bad-noise.toif.txt](https://github.com/trezor/trezor-firmware/files/4952337/bad-noise.toif.txt)
after downloading, rename to `bad-noise.toif`
then:
```
trezorctl set-homescreen -f bad-noise.toif
```

@bosomt can you please test older firmwares if they have the same problem?

matejcik, 2020-07-21:
the problem is that `apply_settings` does not check validity of the image

bosomt, 2020-07-21:
@matejcik 
same behavior with 2.3.0 and 2.3.1 /production versions/
only way to recover was to wipe it including bootloader

bosomt, 2020-08-24:
**QA OK**

now it says

`tomas@tomas:~]$ trezorctl set-homescreen -f ./Downloads/bad-noise.toif
Error: DataError: Invalid homescreen
`

sorooris, 2020-08-27:
**QA OK**

device: model T 2.3.3 df5421e
trezorctl 0.12.1
[image]