trezor/trezor-firmware #1122: Implement new fonts

tsusanka, 2020-07-23:
For T1 (but in Core!) we want to introduce PixelOperator8 (regular and bold) and for TT we want to introduce Hoves (regular, medium and bold). These fonts will completely replace all the current ones.

[fonts.zip](https://github.com/trezor/trezor-firmware/files/4964573/fonts.zip)

prusnak, 2020-07-23:
What about the monospace variant for TT? The font we currently use (Roboto) has both non-proportional (monospace) and proportional variants. TT Hoves does not seem to have one.

tsusanka, 2020-07-23:
Very good points.

> What about the monospace variant for TT? Wwe need this for addresses, etc. The font we currently use (Roboto) has both non-proportional (monospace) and proportional variants. TT Hoves does not seem to have one. Will we stick with Roboto for monospace?

I'll clear that up with Petr and let you know.

> Are there any Trezor screen mockups using the new font? I am not sure how TT Hoves is readable on small displays (while Roboto was designed exactly for this purpose).

All screens are on [zeplin](https://app.zeplin.io/projects?pid=5ec4dfb6824fd12f62639331). I have uploaded [this screen](https://app.zeplin.io/project/5ec4dfb6824fd12f62639331/screen/5eea10f3fa85b246e9c7afb5) to TT via  `git checkout toif-upload`
and `tools/toif_upload image.png` and it seems readable just fine _but_ it is true some of the font is quite small. Since this is an important change we should definitely decide first if this is the way we want to go before doing any code changes. I have scheduled a meeting with Ondra (the other designer) to evaluate this and I will bring it up on future Product meeting as well.

tsusanka, 2020-07-23:
> I'll clear that up with Petr and let you know.

Ok, sorry for the confusion, Petr was inaccurate. We want to add the aforementioned fonts and keep Roboto Mono exactly for addresses, amounts etc. I will edit it above.

prusnak, 2020-07-28:
Added PixelOperator font (for T1) to Trezor Core in https://github.com/trezor/trezor-firmware/commit/47e75504239a988db7c82a871a2bf0075c2393bb

prusnak, 2020-07-28:
Trying new fonts for core in the `new-fonts` branch - it currently fails, but one can see the new screenshots in the `ui tests` artefacts.

tsusanka, 2020-07-31:
Petr's library:

[image]

but we want to discuss this further.

prusnak, 2020-07-31:
Current mapping `new-fonts` branch:
* normal -> TT Hoves Regular 18
* medium -> TT Hoves Medium 20
* bold -> TT Hoves Bold 16
* mono -> Roboto Mono Regular 20

toledot21, 2021-01-27:
Hi, sorry to drop in here, but I thought it wasn't good to open another issue on the same matter. Its just a request:
I was wondering how can I generate 20+ font sizes following your glyph profile from font_roboto_regular_20.c/h, for instance(32, 48 72 font size).
Thank you in advance,
Hope to hear from you,
Regards,

closed 2022-02-10 by mmilata