trezor/trezor-firmware #1125: Add support for Bitcoin Cash SLP tokens

gasull, 2020-07-23:
Please add support for [Simple Ledger Protocol (SLP)](https://simpleledger.cash/) tokens on Bitcon Cash, which are based on `OP_RETURN`.

Some of the most common SLP tokens are:

 - [Tether (USDT)](https://finance.yahoo.com/news/tether-stablecoin-launches-seventh-blockchain-160000007.html),
 - [Honestcoin (USDH)](https://www.honestcoin.io/),
 - [Core Cash (BTC2)](https://bitcoinist.com/btc2-token-levels-bitcoin-fees-with-bitcoin-cash/).

baoduy, 2022-03-01:
Hi, I think supporting USDT and UST is really needed and it should not be prioritised as low because a lot of people trading these coins. Could you please let us know when these will be supported?

Hannsek, 2023-02-09:
I am closing it now as we don't have any plans to implement this in foreseeable future. We can re-open it later.

closed 2023-02-09 by Hannsek