trezor/trezor-firmware #1130: trezorctl: unify option arguments

matejcik, 2020-07-24:
due to the development history, similar subcommands in `trezorctl` have wildly different parameter conventions:

* To enable pin, use `trezorctl set pin`. To disable pin, use `trezorctl set pin -r`
* To enable passphrase, use `trezorctl set passphrase enabled`. To disable, `trezorctl set passphrase disabled`
* To enable SD-protect, use `trezorctl device sd-protect enable`. To disable, use `trezorctl device sd-protect disable`
* some commands accept a single required option, other accept a single argument
* etc.

we should come up with a set of guidelines and make sure all commands conform to them.

grdddj, 2022-03-24:
Commands for `QA` to try:

`trezorctl set pin`
`trezorctl set wipe-code`
`trezorctl set passphrase`
`trezorctl device sd-protect`

Check that all these commands take an arguments `on` and `off` as a way to specify the positive and negative effects (`trezorctl set pin on` vs `trezorctl set pin off`).

In the case of `trezorctl set passphrase`, also the `enabled/disabled` argument should work (to maintain backward compatibility) and be identical to `on/off`.

In the case of `trezorctl set pin` and `trezorctl set wipe-code`, the "legacy" `-r` flag is still supported as a way to "remove" - the same as `off` argument. These two cannot conflict with each other, therefore the usage `trezorctl set pin on -r` should return an error.

wendys-cats, 2022-04-05:
**QA NOK** - commands trezorctl set passphrase is not valid, nor is trezorctl device sd-protect

**Info**
**trezorctl** version 0.13.1 (via poetry from monorepo)
**device** used: TT 2.5.0
**OS:** Nix

**trezorctl device sd-protect**
[image]

**trezorctl set passphrase**
[image]

**List of tried commands:**
 set pin OK
 set pin off OK
set pin on OK
set pin -r OK
set wipe-code OK
set wipe code on OK
set wipe code -r OK
set wipe code off OK
**set passphrase NOK**
set passphrase on OK
set passphrase off OK
set passphrase enabled OK
set passphrase disabled OK
**device sd-protect NOK**
device sd-protect on OK
device sd-protect off OK
device sd-protection refresh OK

grdddj, 2022-04-06:
> commands trezorctl set passphrase is not valid, nor is trezorctl device sd-protect

@wendys-cats With @matejcik I think we agreed that we will not be supporting default arguments in these commands, so without `on/off`, they should really be invalid. If they were really valid before, we can quite easily enable it for backward-compatibility purposes.

matejcik, 2022-04-06:
Correct. I don't think these commands worked without argument before, so this result is OK from my side.

grdddj, 2022-04-06:
> don't think these commands worked without argument before

Right, I tried it on `master` before the change was merged, and none of those worked before:
```
(.venv)  /home/jmusil/trezor-firmware  ➦ e9364ea0a  trezorctl set passphrase                             
Usage: trezorctl set passphrase [OPTIONS] COMMAND [ARGS]...

  Enable, disable or configure passphrase protection.

Options:
  --help  Show this message and exit.

Commands:
  disabled  Disable passphrase.
  enabled   Enable passphrase.
```

```
(.venv)  /home/jmusil/trezor-firmware  ➦ e9364ea0a  trezorctl device sd-protect
Usage: trezorctl device sd-protect [OPTIONS] {enable|disable|refresh}
Try 'trezorctl device sd-protect --help' for help.

Error: Missing argument '{enable|disable|refresh}'. Choose from:
        enable,
        disable,
        refresh
```

wendys-cats, 2022-04-06:
Great, changing to **QA OK** then :)