trezor/trezor-firmware #1131: Firmware spends a lot of time looking up qstrings

hiviah, 2020-07-24:
**Describe the bug**
Signing code (and probably other code) spends abou 33% in `qstr_find_strn`.

**Firmware version and revision**
2.3.1 with synchronous protobuf branch

**To Reproduce**
Use `pytest tests/device_tests/test_msg_signtx_segwit.py::TestMsgSigntxSegwit::test_lots_of_segwit_inputs` test from `tsusanka/streaming-timing` branch. 

Probably any signing test will suffice, I can see the qstring find function when running other signing tests taking 10-25%.

Measurements from ITM trace PC sampling:

Total cumulative time spent in functions when test runs:

        % FUNCTION
    30.46 qstr_find_strn
    15.76 gc_collect_end
     7.54 gc_alloc
     6.45 gc_mark_subtree
     5.62 mp_map_lookup
     4.33 mp_execute_bytecode
     3.76 sha256_Transform
     1.78 mp_obj_get_type
     1.51 find_qstr
     1.43 memset
     0.70 mp_binary_op
     0.65 qstr_hash

Sample of 1 second running signing code:

     33.41%     3473 qstr_find_strn
     17.98%     1870 gc_mark_subtree
      8.77%      912 gc_alloc
      5.90%      614 mp_map_lookup
      4.62%      481 mp_execute_bytecode
      4.20%      437 sha256_Transform
      3.57%      372 gc_sweep
      2.47%      257 gc_deal_with_stack_overflow
      1.87%      195 mp_obj_get_type
      1.64%      171 memset
      1.56%      163 find_qstr
      0.68%       71 mp_setup_code_state
      0.64%       67 mp_binary_op
      0.57%       60 mp_load_method_maybe
      0.56%       59 qstr_hash
    -----------------
     88.44%     9202 of 10395 Samples

     Ex |   Count  |  MaxD | TotalTicks  |  AveTicks  |  minTicks  |  maxTicks 
    ----+----------+-------+-------------+------------+------------+------------
     15 |     1014 |     2 |          0  |          0 |         0  |         0
     93 |     1149 |     1 |          0  |          0 |         0  |         0

    [---H] Interval = 1004mS

prusnak, 2020-07-24:
I just did a small experiment, I patched our copy of the micropython using the following (removing `qstr_find_strn` lookups from `objstr.c`:

```patch
diff --git a/py/objstr.c b/py/objstr.c
index e221982c5..58b3f8e68 100644
--- a/py/objstr.c
+++ b/py/objstr.c
@@ -167,12 +167,6 @@ mp_obj_t mp_obj_str_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_
                 }
                 #endif
 
-                // Check if a qstr with this data already exists
-                qstr q = qstr_find_strn((const char*)str_data, str_len);
-                if (q != MP_QSTRnull) {
-                    return MP_OBJ_NEW_QSTR(q);
-                }
-
                 mp_obj_str_t *o = MP_OBJ_TO_PTR(mp_obj_new_str_copy(type, NULL, str_len));
                 o->data = str_data;
                 o->hash = str_hash;
@@ -2036,16 +2030,6 @@ mp_obj_t mp_obj_new_str_via_qstr(const char* data, size_t len) {
 // the exact length required and then reused for the str/bytes object.  The vstr
 // is cleared and can safely be passed to vstr_free if it was heap allocated.
 mp_obj_t mp_obj_new_str_from_vstr(const mp_obj_type_t *type, vstr_t *vstr) {
-    // if not a bytes object, look if a qstr with this data already exists
-    if (type == &mp_type_str) {
-        qstr q = qstr_find_strn(vstr->buf, vstr->len);
-        if (q != MP_QSTRnull) {
-            vstr_clear(vstr);
-            vstr->alloc = 0;
-            return MP_OBJ_NEW_QSTR(q);
-        }
-    }
-
     // make a new str/bytes object
     mp_obj_str_t *o = m_new_obj(mp_obj_str_t);
     o->base.type = type;
@@ -2063,14 +2047,7 @@ mp_obj_t mp_obj_new_str_from_vstr(const mp_obj_type_t *type, vstr_t *vstr) {
 }
 
 mp_obj_t mp_obj_new_str(const char* data, size_t len) {
-    qstr q = qstr_find_strn(data, len);
-    if (q != MP_QSTRnull) {
-        // qstr with this data already exists
-        return MP_OBJ_NEW_QSTR(q);
-    } else {
-        // no existing qstr, don't make one
-        return mp_obj_new_str_copy(&mp_type_str, (const byte*)data, len);
-    }
+    return mp_obj_new_str_copy(&mp_type_str, (const byte*)data, len);
 }
 
 mp_obj_t mp_obj_str_intern(mp_obj_t str) {
```

I confirmed this boots and works correctly for a very simple scenario (get-features).

@hiviah Could you please try this patch with `test_lots_of_segwit_inputs` and compare the timing and orbtop output before and after applying the patch? Thanks!

matejcik, 2020-07-24:
we could make `trezorutils.decode_bytes(b: bytes, offset: int = None, length: int = None) -> str` function that would simply call, presumably, `mp_obj_new_str_copy`?
and use this in protobuf code

this would have the added advantage that the data could go straight from the USB buffer to the string (Currently we need to allocate an intermediate bytearray. I'm not sure if there is some sort of COW under the hood, but it's a new object allocation in any case.)

prusnak, 2020-07-24:
> we could make `trezorutils.decode_bytes(b: bytes, offset: int = None, length: int = None) -> str` function that would simply call, presumably, `mp_obj_new_str_copy`?
> and use this in protobuf code

This sounds neat!

prusnak, 2020-07-24:
I pushed a new branch: [decode_bytes](https://github.com/trezor/trezor-firmware/tree/decode_bytes) where I implemented @matejcik's idea (without variable offset/length and I only replaced `decode` call only in `protobuf`). Can you @hiviah test that branch as well?

hiviah, 2020-07-24:
With the qstr patch, the lots_of_segwit_inputs test takes 5:22, compared to 7:09 without the patch.

The profile also shows much less time in `qstr_find_strn`:

        % FUNCTION
     8.17 *SLEEP*
    21.39 gc_collect_end
     8.73 gc_mark_subtree
     8.28 gc_alloc
     7.75 mp_map_lookup
     5.99 mp_execute_bytecode
     5.34 sha256_Transform
     3.25 memset
     2.83 bn_multiply_reduce_step
     2.47 mp_obj_get_type
     2.13 find_qstr
     1.42 bn_multiply_long
     1.05 qstr_find_strn
     0.96 mp_binary_op
     0.90 qstr_hash

hiviah, 2020-07-24:
The `decode_bytes` branch seems to be a slower than original - 7:55 instead of 7:09. The `qstr_find_strn` is still called often which suggests that it's called from other parts of code, maybe even micropython's builtin functions:

        % FUNCTION
     9.20 *SLEEP*
    30.34 qstr_find_strn
    14.01 gc_collect_end
     7.07 mp_map_lookup
     5.31 gc_alloc
     4.93 mp_execute_bytecode
     4.65 gc_mark_subtree
     3.71 sha256_Transform
     1.47 frexpf
     1.46 gc_free.part.0
     1.40 qstr_add
     1.08 mp_obj_get_type

hiviah, 2020-07-25:
I put together the qstr patch, removing of gc.collects in bitcoin signing and merged the #1136  unimport branch as a test.

The final time for the `lots_of_segwit_inputs` test is now down to 4:20 (we started at something over 10 min when streaming was first implemented).

However I wonder whether the qstr patch won't make memory pressure or fragmentation worse.

jpochyla, 2020-07-25:
> I pushed a new branch: [decode_bytes](https://github.com/trezor/trezor-firmware/tree/decode_bytes) where I implemented @matejcik's idea (without variable offset/length and I only replaced `decode` call only in `protobuf`). Can you please @hiviah test that branch as well?

I think it's not a good idea to use `mp_obj_new_str_copy` for this, as the backing buffer from the bytearray can be transferred directly to the str, instead of copied and thrown away.

tsusanka, 2021-06-03:
@hiviah will merge this on the #1541 branch.

matejcik, 2021-06-03:
> @hiviah will merge this on the #1541 branch.

what is "this" ? :)

We could apply patch from https://github.com/trezor/trezor-firmware/issues/1131#issuecomment-663534334 experimentally and see if nothing breaks. AFAICT it shouldn't.

hiviah, 2021-06-03:
@matejcik 

Clarification:

Currently with the `matejcik/rust-protobuf-rebased` (commit 84b640ab7ecb464fc57f8ae7d7707efb274860e4) the `qstr_find_strn` goes down to about just 6%. So maybe not worth the hassle having to upkeep the patch above in this issue.

matejcik, 2021-06-03:
Still, 6% is nice.

I posted in the micropython forums to see if we could upstream the patch.

tsusanka, 2021-06-03:
I meant "measure" not "merge", apologies.

hiviah, 2021-06-08:
Updated patch to match current master:

    diff --git a/py/objstr.c b/py/objstr.c
    index a276a255e..56bad7f94 100644
    --- a/py/objstr.c
    +++ b/py/objstr.c
    @@ -167,12 +167,6 @@ mp_obj_t mp_obj_str_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_
                     }
                     #endif
     
    -                // Check if a qstr with this data already exists
    -                qstr q = qstr_find_strn((const char *)str_data, str_len);
    -                if (q != MP_QSTRnull) {
    -                    return MP_OBJ_NEW_QSTR(q);
    -                }
    -
                     mp_obj_str_t *o = MP_OBJ_TO_PTR(mp_obj_new_str_copy(type, NULL, str_len));
                     o->data = str_data;
                     o->hash = str_hash;
    @@ -2055,16 +2049,6 @@ mp_obj_t mp_obj_new_str_via_qstr(const char *data, size_t len) {
     // the exact length required and then reused for the str/bytes object.  The vstr
     // is cleared and can safely be passed to vstr_free if it was heap allocated.
     mp_obj_t mp_obj_new_str_from_vstr(const mp_obj_type_t *type, vstr_t *vstr) {
    -    // if not a bytes object, look if a qstr with this data already exists
    -    if (type == &mp_type_str) {
    -        qstr q = qstr_find_strn(vstr->buf, vstr->len);
    -        if (q != MP_QSTRnull) {
    -            vstr_clear(vstr);
    -            vstr->alloc = 0;
    -            return MP_OBJ_NEW_QSTR(q);
    -        }
    -    }
    -
         // make a new str/bytes object
         mp_obj_str_t *o = m_new_obj(mp_obj_str_t);
         o->base.type = type;
    @@ -2082,14 +2066,7 @@ mp_obj_t mp_obj_new_str_from_vstr(const mp_obj_type_t *type, vstr_t *vstr) {
     }
     
     mp_obj_t mp_obj_new_str(const char *data, size_t len) {
    -    qstr q = qstr_find_strn(data, len);
    -    if (q != MP_QSTRnull) {
    -        // qstr with this data already exists
    -        return MP_OBJ_NEW_QSTR(q);
    -    } else {
    -        // no existing qstr, don't make one
    -        return mp_obj_new_str_copy(&mp_type_str, (const byte *)data, len);
    -    }
    +    return mp_obj_new_str_copy(&mp_type_str, (const byte *)data, len);
     }
     
     mp_obj_t mp_obj_str_intern(mp_obj_t str) {

hiviah, 2021-06-08:
Code profile without and with patch on current master:

Without patch:
[image]

With patch:
[image]

The patch about halves the time in qstr_find_strn.

matejcik, 2021-06-09:
Please push the patch to some branch in our micropython repo. We should experiment with this more -- istm even the 4.6 % is an order of magnitude more than this function should be called. Presumably the profile also shows callers?

hiviah, 2021-06-09:
The qstr patch is in `qstr_patch` branch in micropython repo now.

matejcik, 2021-07-21:
@hiviah can you summarize your latest findings (qstr usage in imports)? i have a vague notion that we can close this...?

hiviah, 2021-08-02:
Yes, we can close this, summary is that we could use the `qstr_patch` above. As to (re-)imports (which are hardwired to use qstrings) is that there is no practical way to avoid them (without some wild changes in VM).

closed 2021-08-02 by hiviah