trezor/trezor-firmware #1132: Don't wrap UI operations in unimports when bitcoin signing

mmilata, 2020-07-24:
Follow up of #1108. The unimports are _probably_ not needed and make the code less readable.

Question: can removing the unimports cause the firmware to run out of memory? Is running the device tests enough to check that it doesn't happen?

matejcik, 2020-07-24:
It could in theory, but I believe we have large enough margins for Bitcoin signing in particular.

The best way to figure out if that doesn't happen is to run some tests with the white-noise homescreen set. @bosomt is skilled at that .)
Perhaps we could make a special FW build where the white-noise homescreen is the default? So that this can run the device-tests suite?

closed 2020-07-28 by tsusanka

sorooris, 2020-08-25:
Blocked by #1222

bosomt, 2020-08-31:
just sent this TEST 10k size + transaction + whitenoise background with 2.3.3 rev df5421e
https://tbtc1.trezor.io/tx/2150d8402380cca111f59c01d053023df7697d8e749729efc64425601fc7331b

@tsusanka @matejcik is that enough ? /+ is this model T specific ? /

matejcik, 2020-08-31:
yes and yes

sorooris, 2020-09-01:
**QA OK**
Device: 2.3.3 df5421e + white noise homescreen
Suite [3095a174f258c0226faf2ebcbdda3d8b7f04686f](https://github.com/trezor/trezor-suite/commits/3095a174f258c0226faf2ebcbdda3d8b7f04686f)

PR #1136 (2020-07-28): core: remove unimports from bitcoin sign_tx layouts
Fixes #1132. Seems to speed up `TestMsgSigntx::test_lots_of_inputs` on device further ~25 % compared to #1128.

Tested with `tests/device_tests/test_msg_signtx{,_external,_mixed_inputs,_prevhash,_segwit,_segwit_native}.py` though I haven't been able to set up the white noise homescreen.